dusty-nv/jetson-inference #388: Jetson TX2 Inference measurement

SusmithaGadagottu, 2019-08-15:
Hi,

I am new to the DeepLearning area. I have the Jetson TX2 board purchased. I would like to run the inference applications for Image classification and object detection on Jetson TX2 board. I have set up the system, installed Jetpack etc.
I am trying to measure the throughput, accuracy, latency and confidence rate etc.

For image classification I have executed the program as mentioned in the instructions:
https://github.com/dusty-nv/jetson-inference/blob/master/docs/imagenet-console-2.md#using-the-console-program-on-jetson

When i executed the program i see below results:
device GPU, networks/ResNet-50/ResNet-50.caffemodel initialized.
[TRT]   networks/ResNet-50/ResNet-50.caffemodel loaded
imageNet -- loaded 1000 class info entries
networks/ResNet-50/ResNet-50.caffemodel initialized.
[image] loaded 'orange_0.jpg'  (1920 x 1920, 3 channels)
class 0950 - 0.998047  (orange)
imagenet-console:  'orange_0.jpg' -> 99.80469% class #950 (orange)

[TRT]   ----------------------------------------------
[TRT]   Timing Report networks/ResNet-50/ResNet-50.caffemodel
[TRT]   ----------------------------------------------
[TRT]   Pre-Process   CPU  0.07712ms  CUDA  0.48864ms
[TRT]   Network       CPU 105.64045ms  CUDA 104.41174ms
[TRT]   Post-Process  CPU  1.95469ms  CUDA  2.86877ms
[TRT]   Total         CPU 107.67226ms  CUDA 107.76915ms
[TRT]   ----------------------------------------------

From the above results, it mentioned that the device is GPU. How do i know if these results are from TensorRT accelerator chip. How can I measure throughput, Latency and accuracy or mAP from these results for the TensorRT accelerator?

Could someone please clarify urgently?

Thanks
Susmitha

dusty-nv, 2019-08-16:
Hi Susmitha, for benchmarking you probably want to use the trt-exec program, it is included with the TensorRT samples under /usr/src/tensorrt

https://github.com/dusty-nv/jetson-inference/blob/master/docs/imagenet-console-2.md#using-the-console-program-on-jetson

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

SusmithaGadagottu, 2019-08-16:
ok. Thanks.
Can you point me to some documentation? 

Presently I would like to run YoloV2, Resnet50, VGG16 and if possible pose estimation models. Also I would want to make sure that the image is classified or objects are detected correctly along with the Latency, accuracy and throughput calculation. Can I run and get these results from trt-exec program? 

Would you be able to give me some guidance?

Thanks
Susmitha

SusmithaGadagottu, 2019-08-16:
I ran the trtexec with the supported parameters. But below arguments are not supported when I executed.
--useSpinWait --allowGPUFallback --useDLACore=0

I get below error:
Unknown argument: --useSpinWait
Unknown argument: --useSpinWait
Unknown argument: --useDLACore

I ran this on TX2. Do you know why I see these errors?

Thanks
Susmitha

dusty-nv, 2019-08-16:
Those are for Xavier, you shouldn’t need them on TX2.

SusmithaGadagottu, 2019-08-16:
i ran the command on TX2 without --useSpinWait --allowGPUFallback --useDLACore=0 and i see below result. Not sue what to analyze from this result. Is it printing the latency on the chip? If I have to run the tests on some datasets can I still use trtexec program?

./trtexec --deploy='/home/nvidia/jetson-inference/data/networks/ResNet-50/deploy.prototxt' --output=stage4_unit3_conv3 --model='/home/nvidia/jetson-inference/data/networks/ResNet-50/ResNet-50.caffemodel' --batch=2 --iterations=10 --avgRuns=100 --fp16 --verbose 

deploy: /home/nvidia/jetson-inference/data/networks/ResNet-50/deploy.prototxt

output: stage4_unit3_conv3

model: /home/nvidia/jetson-inference/data/networks/ResNet-50/ResNet-50.caffemodel
batch: 2
iterations: 10
avgRuns: 100
fp16
verbose
Input "data": 3x224x224
Output "stage4_unit3_conv3": 2048x7x7
Original: 126 layers
After dead-layer removal: 121 layers
After scale fusion: 121 layers
Fusing conv0_1 with relu0_1
Fusing conv0_2 with relu0_2
Fusing conv0_3 with relu0_3
Fusing stage1_unit1_conv1 with stage1_unit1_relu1
Fusing stage1_unit1_conv2 with stage1_unit1_relu2
Fusing stage1_unit1_conv3 with _plus0
Fusing stage1_unit1_conv3 + _plus0 with stage1_unit1_relu3
Fusing stage1_unit2_conv1 with stage1_unit2_relu1
Fusing stage1_unit2_conv2 with stage1_unit2_relu2
Fusing stage1_unit2_conv3 with _plus1
Fusing stage1_unit2_conv3 + _plus1 with stage1_unit2_relu3
Fusing stage1_unit3_conv1 with stage1_unit3_relu1
Fusing stage1_unit3_conv2 with stage1_unit3_relu2
Fusing stage1_unit3_conv3 with _plus2
Fusing stage1_unit3_conv3 + _plus2 with stage1_unit3_relu3
Fusing stage2_unit1_conv1sc with _plus3
Fusing stage2_unit1_conv1 with stage2_unit1_relu1
Fusing stage2_unit1_conv2 with stage2_unit1_relu2
Fusing stage2_unit1_conv1sc + _plus3 with stage2_unit1_relu3
Fusing stage2_unit2_conv1 with stage2_unit2_relu1
Fusing stage2_unit2_conv2 with stage2_unit2_relu2
Fusing stage2_unit2_conv3 with _plus4
Fusing stage2_unit2_conv3 + _plus4 with stage2_unit2_relu3
Fusing stage2_unit3_conv1 with stage2_unit3_relu1
Fusing stage2_unit3_conv2 with stage2_unit3_relu2
Fusing stage2_unit3_conv3 with _plus5
Fusing stage2_unit3_conv3 + _plus5 with stage2_unit3_relu3
Fusing stage2_unit4_conv1 with stage2_unit4_relu1
Fusing stage2_unit4_conv2 with stage2_unit4_relu2
Fusing stage2_unit4_conv3 with _plus6
Fusing stage2_unit4_conv3 + _plus6 with stage2_unit4_relu3
Fusing stage3_unit1_conv1sc with _plus7
Fusing stage3_unit1_conv1 with stage3_unit1_relu1
Fusing stage3_unit1_conv2 with stage3_unit1_relu2
Fusing stage3_unit1_conv1sc + _plus7 with stage3_unit1_relu3
Fusing stage3_unit2_conv1 with stage3_unit2_relu1
Fusing stage3_unit2_conv2 with stage3_unit2_relu2
Fusing stage3_unit2_conv3 with _plus8
Fusing stage3_unit2_conv3 + _plus8 with stage3_unit2_relu3
Fusing stage3_unit3_conv1 with stage3_unit3_relu1
Fusing stage3_unit3_conv2 with stage3_unit3_relu2
Fusing stage3_unit3_conv3 with _plus9
Fusing stage3_unit3_conv3 + _plus9 with stage3_unit3_relu3
Fusing stage3_unit4_conv1 with stage3_unit4_relu1
Fusing stage3_unit4_conv2 with stage3_unit4_relu2
Fusing stage3_unit4_conv3 with _plus10
Fusing stage3_unit4_conv3 + _plus10 with stage3_unit4_relu3
Fusing stage3_unit5_conv1 with stage3_unit5_relu1
Fusing stage3_unit5_conv2 with stage3_unit5_relu2
Fusing stage3_unit5_conv3 with _plus11
Fusing stage3_unit5_conv3 + _plus11 with stage3_unit5_relu3
Fusing stage3_unit6_conv1 with stage3_unit6_relu1
Fusing stage3_unit6_conv2 with stage3_unit6_relu2
Fusing stage3_unit6_conv3 with _plus12
Fusing stage3_unit6_conv3 + _plus12 with stage3_unit6_relu3
Fusing stage4_unit1_conv1sc with _plus13
Fusing stage4_unit1_conv1 with stage4_unit1_relu1
Fusing stage4_unit1_conv2 with stage4_unit1_relu2
Fusing stage4_unit1_conv1sc + _plus13 with stage4_unit1_relu3
Fusing stage4_unit2_conv1 with stage4_unit2_relu1
Fusing stage4_unit2_conv2 with stage4_unit2_relu2
Fusing stage4_unit2_conv3 with _plus14
Fusing stage4_unit2_conv3 + _plus14 with stage4_unit2_relu3
Fusing stage4_unit3_conv1 with stage4_unit3_relu1
Fusing stage4_unit3_conv2 with stage4_unit3_relu2
After vertical fusions: 56 layers
After swap: 56 layers
After final dead-layer removal: 56 layers
After tensor merging: 56 layers
After concat removal: 56 layers
Graph costruction and optimization completed in 0.012156 seconds.
Killed

SusmithaGadagottu, 2019-08-22:
[ImageClassificationExample_onTX2_Novumindresnet50.txt](https://github.com/dusty-nv/jetson-inference/files/3532279/ImageClassificationExample_onTX2_Novumindresnet50.txt)

Can you please let me know if the displayed results are from GPU or TensorRT?

dusty-nv, 2019-08-23:
GPU time is the GPU time taken by TensorRT.
The first run will be slow, and don't measure benchmark results from just one run (see the python examples which do multiple runs with the --profile flag).  Recommend running 'sudo jetson_clocks' before.

SusmithaGadagottu, 2019-08-23:
Thank you so much for your help!!

closed 2019-08-23 by SusmithaGadagottu