dusty-nv/jetson-inference #391: Build From Source Fails. jetson-inference.so not found

MauiJerry, 2019-08-18:
Installing on new Xavier, took a while to get thru the system install, then turned to Hello AI World to setup and try dev environment.  Alas once I followed[ the instructions ](https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md) I tried the tests ...
'''time ./imagenet-console.py --network=googlenet orange_0.jpg output_0.jpg
jetson.inference.__init__.py
Traceback (most recent call last):
  File "./imagenet-console.py", line 24, in <module>
    import jetson.inference
  File "/usr/lib/python2.7/dist-packages/jetson/inference/__init__.py", line 4, in <module>
    from jetson_inference_python import *
ImportError: libjetson-utils.so: cannot open shared object file: No such file or directory
'''
Ok, so maybe python3?... nope
'''
python3 ./imagenet-console.py --network=googlenet orange_0.jpg output_0.jpg
jetson.inference.__init__.py
Segmentation fault (core dumped)
nvidia@xavier:~/Desktop/jetson-inference/build/aarch64/bin$ python3 --version
Python 3.6.8
'''
Digging back, the CMakePreBuild.sh runs the model+pytorch installers which clears the scroll back so I couldnt follow that back further.  When I did edit it to Break on Errors and not run installers I got...
'''
sudo apt-get install -y libopencv-calib3d-dev libopencv-dev 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libopencv-dev is already the newest version (3.3.1-2-g31ccdfe11).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libopencv-calib3d-dev : Depends: libopencv-features2d-dev (= 3.2.0+dfsg-4ubuntu0.1)
 libopencv-dev : Conflicts: libopencv-calib3d-dev
E: Unable to correct problems, you have held broken packages.
'''
trying to install features2d-dev wound up with more broken packages messages. at which point I stop and post here.

I'd really like to use this beast, but why do the basic installs fail?

MauiJerry, 2019-08-18:
Seeing that note about opencv3.3 I removed it and the whole jetson-infrerence tree and started again.
It just finished running the imagenet-console.py orange example.
I guess somewhere in the intro pages https://elinux.org/Jetson_Zoo  before I got to Hello it had installed opencv.

Either a warning or test for 3.3 or upgrading jetson-inference is needed.

dusty-nv, 2019-08-19:
Hi @MauiJerry, that isn't actually an error if openCV install "fails", it just means that you already have openCV installed.

The Python error you were getting is most commonly seen when `sudo make install` isn't run, because that is the command that installs the Python extension modules for `jetson-inference`.

timvg80, 2019-08-26:
Hi @dusty-nv, I was wondering if you could help me with a similar problem that I am having here please. I have done a clean install of the SD Image onto my Nano, and followed all of the installation instructions line by line, but I am getting the `Segmentation fault (core dumped)` error when trying to run the python command for `imagenet-console.py`. In fact, I am getting this error as soon as the command `import jetson.inference` is run, as I have tried this in the Python REPL. This error is also happening when I run `import jetson.utils` as well.

I think that the installation of the networks must have been fine, as the C++ version of the code is working fine. Is there anything that you could suggest that I try?

timvg80, 2019-08-27:
Hi @dusty-nv, thank you for your reply! The `image-console.py` script is now working fine! Thanks again, for this solution, and for the whole project that you have put together!

gadams999, 2019-09-17:
> @timvg80, try a `sudo make install` followed by a `sudo ldconfig`

For me it was the `sudo ldconfig` that resolved the segmentation fault. Thanks!

closed 2023-04-27 by dusty-nv