dusty-nv/jetson-inference #393: Using udpsink or appsink with for detectnet-camera

aviontics, 2019-08-20:
Unsure this is an issue or support request.

I would like to be able to add appsink or udpsink for the gstcamera in ./detectnet-camera so I can stream data to udp port.

Where can I make this change. I looked into the gstCamera files. There were some clues but unsure how to make it work. 

Please help!!!!

dusty-nv, 2019-08-20:
Do you want to do the inferencing also, or do you just want to stream a camera over the network?   See the L4T Accelerated Gstreamer User Guide for example camera pipelines that you can run independently.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

aviontics, 2019-08-20:
Dusty-nv
Thanks for the fast response. 

I want to use it with detectnet-camera code, which I was able to make work with 2 cameras and successfully converted data to opencv mat, for further rprocessing.

I want to be able to stream using udpsink/appsink after processing the image data.

aviontics, 2019-08-21:
Got it working with video writer. 

VideoWriter videoout("appsrc ! videoconvert ! video/x-raw,format=YUY2,width=640,height=480,framerate=30/1 ! jpegenc ! rtpjpegpay ! udpsink host=192.168.0.5 port=5600",CAP_GSTREAMER,0,30,Size(640,480),true);

Seems to automatically get data from the opened camera from gstCamera.

Next is to output opencv transformed image instead of just feed from camera.

dusty-nv, 2019-08-21:
OK, glad you got it working.  Thanks Cromwell.

closed 2023-04-27 by dusty-nv