dusty-nv/jetson-inference #396: Having problem integrating the imagenet classification into an existing pipeline

Source82, 2019-08-21:
Please help I have been trying to do something as simple as converting an image read in using opencv imread, convert it to 1d rgba and make inference to no avail. My code is written below, what am I doing wrong please?
   
    //i have already loaded the model

   //im is a cv Mat BGR image loaded using an existing code
   im = cv::imread('....');

   //here I convert the BGR to RGBA mat
   cv::Mat converted;
   cv::cvtColor(im, converted, cv::COLOR_BGR2RGBA);

   //to convert it into 1d array
   float* convertedIm = new float(im.size[0]* im.size[1]* im.channels());

   if (im.isContinuous()){
          convertedIm2 = (float*)im.data;

   //then I call the imageNet classify function
   float confidence = 0.0f;
   const int img_class = net->Classify(convertedIm, im.rows, im.cols &confidence);

I get this error 
[TRT]   engine.cpp (555) - Cuda Error in execute: 4
[TRT]   engine.cpp (555) - Cuda Error in execute: 4
[TRT]   imageNet::Process() -- failed to execute TensorRT network
[TRT]   imageNet::Process() failed
Feature extraction DONE!
[TRT]   runtime.cpp (30) - Cuda Error in free: 4
terminate called after throwing an instance of 'nvinfer1::CudaError'
  what():  std::exception
Aborted (core dumped)

dusty-nv, 2019-08-21:
Hi @Source82, it would not appear that the `im.data` pointer is in GPU memory, and the `imageNet::Classify()` function expects a CUDA pointer in GPU memory. 

What you can try doing, is allocating a CUDA buffer and copying your image into the CUDA buffer:

``` cpp
#include "cudaMappedMemory.h"

const size_t imgSize = im.size[0]* im.size[1]* im.channels() * sizeof(float)

float* imgCPU  = NULL;
float* imgCUDA = NULL;

if( !cudaAllocMapped((void**)&imgCPU, (void**)&imgCUDA, imgSize) )
{
     printf("failed to allocate CUDA memory for image (%zu bytes)\n", imgSize);
     return 0;
}

memcpy(imgCPU, im.data, imgSize);

float confidence = 0.0f;
const int img_class = net->Classify(imgGPU, im.rows, im.cols, &confidence);
```
 Also it is unclear if `cv::cvtColor()` is converting the image to floating-point.  You will want to make sure it is in float32 RGBA format.

Source82, 2019-08-22:
Thanks it worked!
I first made sure my RGBA image was converted to float just as suggested.

closed 2019-08-22 by Source82

dataplayer12, 2020-11-30:
@dusty-nv Does this also work for engines compiled for DLA? I followed the procedure and the model works perfectly when it runs on GPU but does not detect any objects when the engine is run on DLA0 on Xavier NX. Are there any special considerations when working with DLAs?

dusty-nv, 2020-11-30:
Hi @dataplayer12 , I am not sure I have tested this program on DLA before.   Which model are you running?   

Are you bringing your data in via OpenCV like above?   If so, does DLA work without OpenCV being used?

dataplayer12, 2020-11-30:
I am running `ssd-mobilenet-v2`. I modified `Create` function in `detectnet.h` and specified default device as `DEVICE_DLA`. Data is brought in exactly as you have specified above. The original function with `DEVICE_GPU` works perfectly well and when the device is changed to DLA (with no other changes), it doesn't detect objects. I have not tried running on DLA without using opencv. I will try it and report back.

dataplayer12, 2020-12-03:
@dusty-nv I checked and the problem with DLA is present even when opencv is not used at all. Since the problem extends beyond opencv, I have created a new issue #827 with more information on other networks. Can you please take a look?