dusty-nv/jetson-inference #397: tensorrt inference for yolov3-ssp.weight  for error

molyswu, 2019-08-22:
Try to inference for yolov3-ssp.weight with  tensorrt  on jeson nano ,cuda10.0 ,tensorrt5.1.5,
cd deepstream_reference_apps/yolo/
$trt-yolo-app --flagfile=config/yolov3-ssp.txt

error:

lagfile=config/yolov3-1.txt
Loading pre-trained weights...
Loading complete!
Total Number of weights read : 63052381
      layer               inp_size            out_size       weightPtr
(1)   conv-bn-leaky     3 x 608 x 608      32 x 608 x 608    992   
(2)   conv-bn-leaky    32 x 608 x 608      64 x 304 x 304    19680 
(3)   conv-bn-leaky    64 x 304 x 304      32 x 304 x 304    21856 
(4)   conv-bn-leaky    32 x 304 x 304      64 x 304 x 304    40544 
(5)   skip             64 x 304 x 304      64 x 304 x 304        - 
(6)   conv-bn-leaky    64 x 304 x 304     128 x 152 x 152    114784
(7)   conv-bn-leaky   128 x 152 x 152      64 x 152 x 152    123232
(8)   conv-bn-leaky    64 x 152 x 152     128 x 152 x 152    197472
(9)   skip            128 x 152 x 152     128 x 152 x 152        - 
(10)  conv-bn-leaky   128 x 152 x 152      64 x 152 x 152    205920
(11)  conv-bn-leaky    64 x 152 x 152     128 x 152 x 152    280160
(12)  skip            128 x 152 x 152     128 x 152 x 152        - 
(13)  conv-bn-leaky   128 x 152 x 152     256 x  76 x  76    576096
(14)  conv-bn-leaky   256 x  76 x  76     128 x  76 x  76    609376
(15)  conv-bn-leaky   128 x  76 x  76     256 x  76 x  76    905312
(16)  skip            256 x  76 x  76     256 x  76 x  76        - 
(17)  conv-bn-leaky   256 x  76 x  76     128 x  76 x  76    938592
(18)  conv-bn-leaky   128 x  76 x  76     256 x  76 x  76    1234528
(19)  skip            256 x  76 x  76     256 x  76 x  76        - 
(20)  conv-bn-leaky   256 x  76 x  76     128 x  76 x  76    1267808
(21)  conv-bn-leaky   128 x  76 x  76     256 x  76 x  76    1563744
(22)  skip            256 x  76 x  76     256 x  76 x  76        - 
(23)  conv-bn-leaky   256 x  76 x  76     128 x  76 x  76    1597024
(24)  conv-bn-leaky   128 x  76 x  76     256 x  76 x  76    1892960
(25)  skip            256 x  76 x  76     256 x  76 x  76        - 
(26)  conv-bn-leaky   256 x  76 x  76     128 x  76 x  76    1926240
(27)  conv-bn-leaky   128 x  76 x  76     256 x  76 x  76    2222176
(28)  skip            256 x  76 x  76     256 x  76 x  76        - 
(29)  conv-bn-leaky   256 x  76 x  76     128 x  76 x  76    2255456
(30)  conv-bn-leaky   128 x  76 x  76     256 x  76 x  76    2551392
(31)  skip            256 x  76 x  76     256 x  76 x  76        - 
(32)  conv-bn-leaky   256 x  76 x  76     128 x  76 x  76    2584672
(33)  conv-bn-leaky   128 x  76 x  76     256 x  76 x  76    2880608
(34)  skip            256 x  76 x  76     256 x  76 x  76        - 
(35)  conv-bn-leaky   256 x  76 x  76     128 x  76 x  76    2913888
(36)  conv-bn-leaky   128 x  76 x  76     256 x  76 x  76    3209824
(37)  skip            256 x  76 x  76     256 x  76 x  76        - 
(38)  conv-bn-leaky   256 x  76 x  76     512 x  38 x  38    4391520
(39)  conv-bn-leaky   512 x  38 x  38     256 x  38 x  38    4523616
(40)  conv-bn-leaky   256 x  38 x  38     512 x  38 x  38    5705312
(41)  skip            512 x  38 x  38     512 x  38 x  38        - 
(42)  conv-bn-leaky   512 x  38 x  38     256 x  38 x  38    5837408
(43)  conv-bn-leaky   256 x  38 x  38     512 x  38 x  38    7019104
(44)  skip            512 x  38 x  38     512 x  38 x  38        - 
(45)  conv-bn-leaky   512 x  38 x  38     256 x  38 x  38    7151200
(46)  conv-bn-leaky   256 x  38 x  38     512 x  38 x  38    8332896
(47)  skip            512 x  38 x  38     512 x  38 x  38        - 
(48)  conv-bn-leaky   512 x  38 x  38     256 x  38 x  38    8464992
(49)  conv-bn-leaky   256 x  38 x  38     512 x  38 x  38    9646688
(50)  skip            512 x  38 x  38     512 x  38 x  38        - 
(51)  conv-bn-leaky   512 x  38 x  38     256 x  38 x  38    9778784
(52)  conv-bn-leaky   256 x  38 x  38     512 x  38 x  38    10960480
(53)  skip            512 x  38 x  38     512 x  38 x  38        - 
(54)  conv-bn-leaky   512 x  38 x  38     256 x  38 x  38    11092576
(55)  conv-bn-leaky   256 x  38 x  38     512 x  38 x  38    12274272
(56)  skip            512 x  38 x  38     512 x  38 x  38        - 
(57)  conv-bn-leaky   512 x  38 x  38     256 x  38 x  38    12406368
(58)  conv-bn-leaky   256 x  38 x  38     512 x  38 x  38    13588064
(59)  skip            512 x  38 x  38     512 x  38 x  38        - 
(60)  conv-bn-leaky   512 x  38 x  38     256 x  38 x  38    13720160
(61)  conv-bn-leaky   256 x  38 x  38     512 x  38 x  38    14901856
(62)  skip            512 x  38 x  38     512 x  38 x  38        - 
(63)  conv-bn-leaky   512 x  38 x  38    1024 x  19 x  19    19624544
(64)  conv-bn-leaky  1024 x  19 x  19     512 x  19 x  19    20150880
(65)  conv-bn-leaky   512 x  19 x  19    1024 x  19 x  19    24873568
(66)  skip           1024 x  19 x  19    1024 x  19 x  19        - 
(67)  conv-bn-leaky  1024 x  19 x  19     512 x  19 x  19    25399904
(68)  conv-bn-leaky   512 x  19 x  19    1024 x  19 x  19    30122592
(69)  skip           1024 x  19 x  19    1024 x  19 x  19        - 
(70)  conv-bn-leaky  1024 x  19 x  19     512 x  19 x  19    30648928
(71)  conv-bn-leaky   512 x  19 x  19    1024 x  19 x  19    35371616
(72)  skip           1024 x  19 x  19    1024 x  19 x  19        - 
(73)  conv-bn-leaky  1024 x  19 x  19     512 x  19 x  19    35897952
(74)  conv-bn-leaky   512 x  19 x  19    1024 x  19 x  19    40620640
(75)  skip           1024 x  19 x  19    1024 x  19 x  19        - 
(76)  conv-bn-leaky  1024 x  19 x  19     512 x  19 x  19    41146976
(77)  conv-bn-leaky   512 x  19 x  19    1024 x  19 x  19    45869664
(78)  conv-bn-leaky  1024 x  19 x  19     512 x  19 x  19    46396000
(79)  maxpool         512 x  19 x  19     512 x  15 x  15    46396000
(80)  route                  -            512 x  19 x  19    46396000
(81)  maxpool         512 x  19 x  19     512 x  11 x  11    46396000
(82)  route                  -            512 x  19 x  19    46396000
(83)  maxpool         512 x  19 x  19     512 x   7 x   7    46396000
ERROR: route_83: all concat input tensors must have the same dimensions except on the concatenation axis
trt-yolo-app: /home/ai/deepstream_reference_apps/yolo/lib/trt_utils.cpp:302: std::__cxx11::string dimsToString(nvinfer1::Dims): Assertion `d.nbDims >= 1' failed.
Aborted (core dumped)

Thanks for your help!

molyswu, 2019-08-22:
ry to fix lib/yolo.cpp :385
if ((m_configBlocks.at(i).at("size") == "2" && m_configBlocks.at(i).at("stride") == "1")||(m_configBlocks.at(i).at("size") == "9" && m_configBlocks.at(i).at("stride") == "1")||(m_configBlocks.at(i).at("size") == "5" && m_configBlocks.at(i).at("stride") == "1")||(m_configBlocks.at(i).at("size") == "13" && m_configBlocks.at(i).at("stride") == "1"))

$cd app/trt-yolo/build
$cmake -D CMAKE_BUILD_TYPE=Release .. && make &&make install &&cd ../../..
$trt-yolo-app --flagfile=config/yolov3-spp.txt

error :

Loading pre-trained weights...
Loading complete!
Total Number of weights read : 11114504
layer inp_size out_size weightPtr
(1) conv-bn-leaky 3 x 608 x 608 9 x 608 x 608 279
(2) conv-bn-leaky 9 x 608 x 608 50 x 304 x 304 4529
(3) conv-bn-leaky 50 x 304 x 304 25 x 304 x 304 5879
(4) conv-bn-leaky 25 x 304 x 304 50 x 304 x 304 17329
(5) skip 50 x 304 x 304 50 x 304 x 304 -
(6) conv-bn-leaky 50 x 304 x 304 117 x 152 x 152 70447
(7) conv-bn-leaky 117 x 152 x 152 16 x 152 x 152 72383
(8) conv-bn-leaky 16 x 152 x 152 117 x 152 x 152 89699
(9) skip 117 x 152 x 152 117 x 152 x 152 -
(10) conv-bn-leaky 117 x 152 x 152 40 x 152 x 152 94539
(11) conv-bn-leaky 40 x 152 x 152 117 x 152 x 152 137127
(12) skip 117 x 152 x 152 117 x 152 x 152 -
(13) conv-bn-leaky 117 x 152 x 152 244 x 76 x 76 395035
(14) conv-bn-leaky 244 x 76 x 76 43 x 76 x 76 405699
(15) conv-bn-leaky 43 x 76 x 76 244 x 76 x 76 501103
(16) skip 244 x 76 x 76 244 x 76 x 76 -
(17) conv-bn-leaky 244 x 76 x 76 71 x 76 x 76 518711
(18) conv-bn-leaky 71 x 76 x 76 244 x 76 x 76 675603
(19) skip 244 x 76 x 76 244 x 76 x 76 -
(20) conv-bn-leaky 244 x 76 x 76 74 x 76 x 76 693955
(21) conv-bn-leaky 74 x 76 x 76 244 x 76 x 76 857435
(22) skip 244 x 76 x 76 244 x 76 x 76 -
(23) conv-bn-leaky 244 x 76 x 76 63 x 76 x 76 873059
(24) conv-bn-leaky 63 x 76 x 76 244 x 76 x 76 1012383
(25) skip 244 x 76 x 76 244 x 76 x 76 -
(26) conv-bn-leaky 244 x 76 x 76 48 x 76 x 76 1024287
(27) conv-bn-leaky 48 x 76 x 76 244 x 76 x 76 1130671
(28) skip 244 x 76 x 76 244 x 76 x 76 -
(29) conv-bn-leaky 244 x 76 x 76 56 x 76 x 76 1144559
(30) conv-bn-leaky 56 x 76 x 76 244 x 76 x 76 1268511
(31) skip 244 x 76 x 76 244 x 76 x 76 -
(32) conv-bn-leaky 244 x 76 x 76 60 x 76 x 76 1283391
(33) conv-bn-leaky 60 x 76 x 76 244 x 76 x 76 1416127
(34) skip 244 x 76 x 76 244 x 76 x 76 -
(35) conv-bn-leaky 244 x 76 x 76 43 x 76 x 76 1426791
(36) conv-bn-leaky 43 x 76 x 76 244 x 76 x 76 1522195
(37) skip 244 x 76 x 76 244 x 76 x 76 -
(38) conv-bn-leaky 244 x 76 x 76 457 x 38 x 38 2527595
(39) conv-bn-leaky 457 x 38 x 38 89 x 38 x 38 2568624
(40) conv-bn-leaky 89 x 38 x 38 457 x 38 x 38 2936509
(41) skip 457 x 38 x 38 457 x 38 x 38 -
(42) conv-bn-leaky 457 x 38 x 38 74 x 38 x 38 2970623
(43) conv-bn-leaky 74 x 38 x 38 457 x 38 x 38 3276813
(44) skip 457 x 38 x 38 457 x 38 x 38 -
(45) conv-bn-leaky 457 x 38 x 38 69 x 38 x 38 3308622
(46) conv-bn-leaky 69 x 38 x 38 457 x 38 x 38 3594247
(47) skip 457 x 38 x 38 457 x 38 x 38 -
(48) conv-bn-leaky 457 x 38 x 38 87 x 38 x 38 3634354
(49) conv-bn-leaky 87 x 38 x 38 457 x 38 x 38 3994013
(50) skip 457 x 38 x 38 457 x 38 x 38 -
(51) conv-bn-leaky 457 x 38 x 38 72 x 38 x 38 4027205
(52) conv-bn-leaky 72 x 38 x 38 457 x 38 x 38 4325169
(53) skip 457 x 38 x 38 457 x 38 x 38 -
(54) conv-bn-leaky 457 x 38 x 38 63 x 38 x 38 4354212
(55) conv-bn-leaky 63 x 38 x 38 457 x 38 x 38 4615159
(56) skip 457 x 38 x 38 457 x 38 x 38 -
(57) conv-bn-leaky 457 x 38 x 38 38 x 38 x 38 4632677
[... 47 lines cut ...]
(105) conv-bn-leaky 54 x 38 x 38 23 x 38 x 38 10988606
(106) conv-bn-leaky 23 x 38 x 38 157 x 38 x 38 11021733
(107) conv-linear 157 x 38 x 38 27 x 38 x 38 11025999
(108) yolo 27 x 38 x 38 27 x 38 x 38 11025999
(109) route - 23 x 38 x 38 11025999
(110) conv-bn-leaky 23 x 38 x 38 50 x 38 x 38 11027349
(111) upsample 50 x 38 x 38 50 x 76 x 76 -
(112) route - 294 x 76 x 76 11027349
(113) conv-bn-leaky 294 x 76 x 76 33 x 76 x 76 11037183
(114) conv-bn-leaky 33 x 76 x 76 57 x 76 x 76 11054340
(115) conv-bn-leaky 57 x 76 x 76 15 x 76 x 76 11055255
(116) maxpool 15 x 76 x 76 15 x 76 x 76 11055255
(117) route - 15 x 76 x 76 11055255
(118) maxpool 15 x 76 x 76 15 x 76 x 76 11055255
(119) route - 15 x 76 x 76 11055255
(120) maxpool 15 x 76 x 76 15 x 76 x 76 11055255
(121) route - 30 x 76 x 76 11055255
(122) conv-bn-leaky 30 x 76 x 76 44 x 76 x 76 11067311
(123) conv-bn-leaky 44 x 76 x 76 41 x 76 x 76 11069279
(124) conv-bn-leaky 41 x 76 x 76 63 x 76 x 76 11092778
(125) conv-linear 63 x 76 x 76 27 x 76 x 76 11094506
(126) yolo 27 x 76 x 76 27 x 76 x 76 11094506
Number of unused weights left : 19998
trt-yolo-app: /home/ai/deepstream_reference_apps/yolo/lib/yolo.cpp:409: void Yolo::createYOLOEngine(nvinfer1::DataType, Int8EntropyCalibrator*): Assertion `0' failed.
Aborted (core dumped)

dusty-nv, 2019-08-22:
Closing this issue as it is unrelated to this repo, which does not contain YOLO.  Instead, please post to the DeepStream forum:  https://devtalk.nvidia.com/default/board/209/deepstream-sdk/

closed 2019-08-22 by dusty-nv