dusty-nv/jetson-inference #398: Retrieving multiple layer outputs in single forward pass

Source82, 2019-08-22:
Please is it possible to get the output of multiple layers in a single forward pass?  i.e. simultaneously getting the outputs, instead of creating multiple networks.

dusty-nv, 2019-08-22:
Yes, for example detectNet models typically have multiple output layers.

You need to pass a vector of the desired output layer names to `tensorNet::LoadNetwork()`, see here for an example of this being done:   https://github.com/dusty-nv/jetson-inference/blob/99ecf1f41b47b6d4f8fb848963f4c6517cfa82c3/c/detectNet.cpp#L199

And then when you go to process the network, you pass multiple output buffers to TensorRT:
https://github.com/dusty-nv/jetson-inference/blob/99ecf1f41b47b6d4f8fb848963f4c6517cfa82c3/c/detectNet.cpp#L658

Source82, 2019-08-22:
Thanks so much for the timely and accurate response :-)