dusty-nv/jetson-inference #400: Run Object detection application for YoloV2 model

SusmithaGadagottu, 2019-08-26:
Hi,

Can I run your object detection application for YoloV2 model? I have the yoloV2 prototxt file and the caffe model file. Could you please let me know how I can run the object detection with YoloV2 model on TX2?

Thanks
Susmitha

SusmithaGadagottu, 2019-08-30:
Do you have any update for my above request?

Thanks
Susmitha

MuhammadAsadJaved, 2019-10-16:
@SusmithaGadagottu 
 Have you found the solution?  I also have same problem. I have YOLOV3 caffe weights and prototxt file. but don't understand how to run in deep stream on jetson TX2.

dusty-nv, 2019-10-16:
This `jetson-inference` repo uses SSD-Mobilenet/Inception or DetectNet for object detection.   

See here for guide for using custom YOLO model in DeepStream 4.0:   https://docs.nvidia.com/metropolis/deepstream/4.0/Custom_YOLO_Model_in_the_DeepStream_YOLO_App.pdf

For further assistance with DeepStream, please search or post to the DeepStream SDK forum:  https://devtalk.nvidia.com/default/board/209/deepstream-sdk/

closed 2019-10-16 by dusty-nv