dusty-nv/jetson-inference #402: UffParser: Parser error: network/input/Placeholder: Invalid number of Dimensions 0

dongkaizhou, 2019-08-29:
nvidia@nvidia-desktop:~/project/jetson-inference/build/aarch64/bin$ ./segnet-console city6.png city6_2.png --prototxt=NULL --model=networks/SAITE_SEG/saite_seg.uff --labels=networks/SAITE_SEG/cityscapes-labels.txt --colors=networks/SAITE_SEG/cityscapes-deploy-colors.txt --input_blob=data \ --output_blob=score_fr

segNet -- loading segmentation network model from:
       -- prototxt:   NULL
       -- model:      networks/SAITE_SEG/saite_seg.uff
       -- labels:     networks/SAITE_SEG/cityscapes-labels.txt
       -- colors:     networks/SAITE_SEG/cityscapes-deploy-colors.txt
       -- input_blob  'data'
       -- output_blob 'score_fr_21classes'
       -- batch_size  1

[TRT]   TensorRT version 5.1.6
[TRT]   loading NVIDIA plugins...
[TRT]   Plugin Creator registration succeeded - GridAnchor_TRT
[TRT]   Plugin Creator registration succeeded - NMS_TRT
[TRT]   Plugin Creator registration succeeded - Reorg_TRT
[TRT]   Plugin Creator registration succeeded - Region_TRT
[TRT]   Plugin Creator registration succeeded - Clip_TRT
[TRT]   Plugin Creator registration succeeded - LReLU_TRT
[TRT]   Plugin Creator registration succeeded - PriorBox_TRT
[TRT]   Plugin Creator registration succeeded - Normalize_TRT
[TRT]   Plugin Creator registration succeeded - RPROI_TRT
[TRT]   Plugin Creator registration succeeded - BatchedNMS_TRT
[TRT]   completed loading NVIDIA plugins.
[TRT]   detected model format - UFF  (extension '.uff')
[TRT]   desired precision specified for GPU: FASTEST
[TRT]   requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]   native precisions detected for GPU:  FP32, FP16
[TRT]   selecting fastest native precision for GPU:  FP16
[TRT]   attempting to open engine cache file networks/SAITE_SEG/saite_seg.uff.1.1.GPU.FP16.engine
[TRT]   cache file not found, profiling network model on device GPU
[TRT]   device GPU, loading  networks/SAITE_SEG/saite_seg.uff
[TRT]   UffParser: Parser error: network/input/Placeholder: Invalid number of Dimensions 0
[TRT]   failed to parse UFF model 'networks/SAITE_SEG/saite_seg.uff'
[TRT]   device GPU, failed to load networks/SAITE_SEG/saite_seg.uff
segNet -- failed to initialize.
segnet-console:   failed to initialize segnet

dongkaizhou, 2019-08-29:
This is why ? thanks @dusty-nv

dusty-nv, 2019-08-29:
I'm not sure, I am not familiar with this model (presumably from TensorFlow) or what network it is.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

dusty-nv, 2019-08-29:
See here for follow-up:   https://devtalk.nvidia.com/default/topic/1062181/jetson-tx2/jetson-inference-segnet-console-by-uff-uffparser-parser-error-invalid-number-of-dimensions-0/post/5378829/#5378829

dongkaizhou, 2019-08-30:
Thanks,  I used my segNet's input_blob-- 'network/input/Placeholder' -- output_blob 'network/output/ArgMax',the problem is solution.