dusty-nv/jetson-inference #404: AttributeError: module 'jetson.inference' has no attribute 'imageNet'

davidRetana, 2019-09-04:
I am trying to execute ./imagenet-console.py, i am getting the following error:

$ ./imagenet-console.py sample_image.jpg
Traceback (most recent call last):
  File "./imagenet-console.py", line 33, in <module>
    formatter_class=argparse.RawTextHelpFormatter, epilog=jetson.inference.imageNet.Usage())
AttributeError: module 'jetson.inference' has no attribute 'imageNet'

Looking at the directory: /usr/lib/python3.6/dist-packages/jetson/inference only 1 file is show
jetbot@jetbot:/usr/lib/python3.6/dist-packages/jetson/inference$ ls
__init__.py
And $ cat __init__.py output nothing (is a empty file)

Runnnig the same program in c++ works fine
$ ./imagenet-console sample_image.jpg

Any suggestions?
Thanks in advance

dusty-nv, 2019-09-04:
Hi @davidRetana, you might want to try running these commands:

``` bash
$ sudo apt-get install libpython3-dev python3-numpy
$ cd <jetson-inference>/build
$ cmake ../
$ make
$ sudo make install
$ sudo ldconfig
```

davidRetana, 2019-09-04:
The last command outputs:

jetbot@jetbot:~/jetson-inference/build$ sudo ldconfig
/sbin/ldconfig.real: File /usr/local/lib/libjetson-inference.so is empty, not checked.
/sbin/ldconfig.real: File /usr/local/lib/libjetson-utils.so is empty, not checked.

the problem persist..., maybe my system is broken and i need to reinstall sd image

davidRetana, 2019-09-06:
It seems that the problem is with jetbot SD image. i've tried to build the project in jetson nano normal SD image and works without problem