dusty-nv/jetson-inference #405: Does this repo support jet pack 3.1 with TensorRT 2.1, cuDNN 6.0?

ArtlyStyles, 2019-09-04:


dusty-nv, 2019-09-04:
> -----------------------------------------------------------------------------------
> This email message is for the sole use of the intended recipient(s) and may contain
> confidential information.  Any unauthorized review, use, disclosure or distribution
> is prohibited.  If you are not the intended recipient, please contact the sender by
> reply email and destroy all copies of the original message.
> -----------------------------------------------------------------------------------

Hi Artly, try the L4T-R28.1 branch:  https://github.com/dusty-nv/jetson-inference/tree/L4T-R28.1

ArtlyStyles, 2019-09-05:
Is there an example about how to use Tensorflow uff model with this repo?

dusty-nv, 2019-09-05:
I do not believe TensorRT 2.1 supported TensorFlow UFF models, you would need to upgrade to newer version.

closed 2023-04-27 by dusty-nv