dusty-nv/jetson-inference #406: Using Tensorflow uff model

ArtlyStyles, 2019-09-05:
Is there a example about how to use Tensorflow uff model?

dusty-nv, 2019-09-05:
Yes, detectNet class uses UFF models:  SSD-Mobilenet-v1, SSD-Mobilenet-v2, and SSD-Inception-v2

Also see these other examples of converting to UFF:

https://github.com/AastaNV/TRT_object_detection
https://github.com/NVIDIA-AI-IOT/tf_to_trt_image_classification

TensorRT also includes UFF samples as well.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

slanucha, 2019-10-04:
Hello Dustin, I have one issue with using custom uff models.
I had to retrain the SSD Inception model for my own project. Swapping the ssd_inception_v2_coco.uff and ssd_coco_labels.txt works fine but I wanted to make a cleaner solution and use it as a custom model.

I was experimenting with jetson.inference.detectNet argv list argument but it seems like this only supports caffe models. Do you have any suggestions? Modify the library on my own or just stick to the current solution?

dusty-nv, 2019-10-04:
Hi Szymon, one reason that custom UFF aren't loaded from the command line is that they require more info like the input dimensions, and they use a slightly different Create() function for UFF.

I think the easiest way for you to add a new network enum to detectNet for your model.  Search for instances of SSD_MOBILENET_V2 in the detectNet.h/cpp code to see where to add your own.  Then you can load it from the command line with the --network=ssd-mobilenet-v2-custom or whatever you decide to call it.

Hope that helps!

zychen2016, 2019-11-04:
```
In jetston nano
sudo python3 convert_to_uff.py ~/test/TRT_object_detection/model/ssd_mobilenet_v2_coco_2018_03_29/frozen_inference_graph.pb -o hello.uff -O NMS -p /usr/src/tensorrt/samples/sampleUffSSD/config.py
```

Then  use the uff file

```
./detectnet-camera --model=./networks/hello.uff --class_labels=./networks/tmp/ssd_coco_labels.txt
```

```
[TRT] TensorRT version 5.0.6
[TRT] loading NVIDIA plugins...
[TRT] completed loading NVIDIA plugins.
[TRT] detected model format - UFF (extension '.uff')
[TRT] desired precision specified for GPU: FASTEST
[TRT] requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT] native precisions detected for GPU: FP32, FP16
[TRT] selecting fastest native precision for GPU: FP16
[TRT] attempting to open engine cache file ./networks/hello.uff.1.1.GPU.FP16.engine
[TRT] cache file not found, profiling network model on device GPU
[TRT] device GPU, loading /home/jetbot/test/jetson-inference/build/aarch64/bin/ ./networks/hello.uff
[TRT] FeatureExtractor/MobilenetV2/Conv/Relu6: elementwise inputs must have same dimensions or follow broadcast rules (input dimensions were [1,32,150,150] and [1,1,1])
[TRT] FeatureExtractor/MobilenetV2/expanded_conv/depthwise/depthwise: at least three non-batch dimensions are required for input
[TRT] UFFParser: Parser error: FeatureExtractor/MobilenetV2/expanded_conv/depthwise/BatchNorm/batchnorm/mul_1: The input to the Scale Layer is required to have a minimum of 3 dimensions.
[TRT] failed to parse UFF model './networks/hello.uff'
[TRT] device GPU, failed to load ./networks/hello.uff

```
Any people could help me ?

How to get the uff which could be used in detectnet_camera?

Thank you very much!

revrr, 2020-04-12:
I also cant figure how to convert SSD inception v2 frozen graph .pb to .uff

closed 2023-04-27 by dusty-nv