dusty-nv/jetson-inference #407: use tensorrt to accelerate siamRPN network

JensenHJS, 2019-09-05:
Whether tensorrt can accelerate siamRPN network？SiamRPN network is a target tracking algorithm. It includes a Alexnet subnetwork and a RPN subnetwork.  There is only a fewer layers of SiamRPN

dusty-nv, 2019-09-11:
Hi @JensenHJS, you would want to check if all the layers are supported by TensorRT:

https://docs.nvidia.com/deeplearning/sdk/tensorrt-support-matrix/index.html#layers-matrix

GOBish, 2020-06-23:
Hi - I am also trying to get a Siam network for tracking running on the Jetson Xavier Nx. So far I have only been able to run it with pytorch and it runs fairly slowly, only ~4 FPS, even with 99% GPU load. Any suggestions?

JensenHJS, 2020-06-23:
No,I don't solve this problem. I just start learn how to use trt When I raise this question.  I know that somebody has realized trt to accelerate siamrpn. But maybe it is difficult.  I run siamrpn with pytorch in the tx2，it is about 25 fps. maybe you can use other lightweight backbone ， backbone of alexnet

GOBish, 2020-06-24:
Do you have a link to the code you used? I tried SiamRPN with pytorch and am only getting framerates of under 8 FPS.
thank you.

JensenHJS, 2020-06-24:
https://github.com/STVIR/pysot

GOBish, 2020-06-26:
I am getting ~14-18FPS using SiamRPN with the alexnet backbone on a Xavier nx, which is dissapointing. 
Perhaps it is the webcam.

ShirGrinblat, 2021-06-01:
@JensenHJS @GOBish  Any solution? I'm trying to convert [https://github.com/STVIR/pysot](url) into TensorRT without any success.

closed 2023-04-27 by dusty-nv