dusty-nv/jetson-inference #411: Not an issue, more a question

bobogumizelka, 2019-09-10:
In 'Re-training on Cat/Dog Dataset imagenet-console.py example is used with parameters:

imagenet-console.py --model=cat_dog/resnet18.onnx --input_blob=input_0 --output_blob=output_0 --labels=$DATASET/labels.txt $DATASET/test/cat/01.jpg cat.jpg

How is that possible, if there are no 'model', 'input_blob', 'output_blob' in imagenet_console scrip: https://github.com/dusty-nv/jetson-inference/blob/master/python/examples/imagenet-console.py?

dusty-nv, 2019-09-10:
Hi @bobogumizelka, note that the command line arguments are also passed to the `imageNet` class when it is created:

https://github.com/dusty-nv/jetson-inference/blob/99ecf1f41b47b6d4f8fb848963f4c6517cfa82c3/python/examples/imagenet-console.py#L53

These extra command line arguments are parsed inside of `imageNet` implementation for consistency and simplicity of the programs across Python and C++.

bobogumizelka, 2019-09-12:
Thank you very much for a quick answer! I checked it out, and now it's all clear :)

bobogumizelka, 2019-09-20:
Now I'm loading the detectNet like this: det = jetson.inference.detectNet('facenet') 
Is there a way to check is the net is loaded properly? When I pass an image (loaded with opencv, converted with cudaFromNumpy()) I get an exception:
Exception: jetson.inference -- detectNet.Detect() encountered an error classifying the image
Same image passed to:
net = jetson.inference.imageNet('googlenet',
                                ['--model=resnet18.onnx', 
                                 '--input_blob=input_0', 
                                 '--output_blob=output_0', 
                                 '--labels=labels.txt'])
works fine. That's why I would like to check if the network is loaded properly.

dusty-nv, 2019-09-20:
The program would output any errors loading the detectNet model at the time you instantiated the detectNet object.   Can you post your code and terminal log?

You should be passing a 4-channel RGBA image from OpenCV to `cudaFromNumpy()`.  `cudaFromNumpy()` doesn't change the number of dimensions (although it will convert it to float), and detectNet expects a 4-channel image.

bobogumizelka, 2019-09-24:
Yes, it was 3-channel image problem. Thanks!

closed 2019-09-24 by bobogumizelka

MichiBe91, 2021-01-07:
Hello,

I have the problem that I don't know how to load a custom trained model (like in the tutorial videos: Jetson AI Fundamentals - S3E5) with a python script. With using the flags in the console it is no problem. But how can I write the flags directly in a script. Please See my example below:

`
import jetson.inference
import jetson.utils

net = jetson.inference.detectNet("ssd-mobilenet.onnx", ["model=pathToModel/ssd-mobilenet.onnx", "labels=pathToLabels/labels.txt", "input-blob=input_0", "output-cvg=scores", "output-bbox=boxes"])

camera = jetson.utils.videoSource("csi://0")

display = jetson.utils.videoOutput()

while True:
    img = camera.Capture()
    detections = net.Detect(img)
    display.Render(img)
    display.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))`

I hope somebody can help me. Thank you :)

MichiBe91, 2021-01-07:
Hello, 

I found a solution:

`net = jetson.inference.detectNet(argv=["--model=pathToModel/ssd-mobilenet.onnx", "--labels=pathToLabels/labels.txt", "--input-blob=input_0", "--output-cvg=scores", "--output-bbox=boxes"], threshold=0.5)`

luispaulino1989, 2021-02-25:
> Hello,
> 
> I found a solution:
> 
> `net = jetson.inference.detectNet(argv=["--model=pathToModel/ssd-mobilenet.onnx", "--labels=pathToLabels/labels.txt", "--input-blob=input_0", "--output-cvg=scores", "--output-bbox=boxes"], threshold=0.5)`

Hello,

Im trying to do the same, but not work:

net = jetson.inference.detectNet(argv=["--model=/home/luis/jetson-inference/python/training/detection/ssd/models/frutas/ssd-mobilenet.onnx", "--labels=/home/luis/jetson-inference/python/training/detection/ssd/models/frutas/labels.txt", "--input-blob=input_0", "--output-cvg=scores", "--output-bbox=boxes"], threshold=0.5)

Exception: jetson.inference -- detectNet failed to load network

Do  you know what happen?

This model work good in the console.

MichiBe91, 2021-02-25:
Mhh - strange. Maybe you could try it with double slash (//). You could also try to load the model from the same folder as your script. So you can be sure that there is no mistake in the path.

Lyns01, 2021-05-25:
Hi @MichiBe91, I had the same doubt and your answer helped me thank you.

AlperGith, 2021-05-25:
@Lyns01 Hey, i need your help

AlperGith, 2021-05-25:
net = jetson.inference.detectNet(argv=["--model=/jetson-inference/python/training/detection/ssd/models/Khas/ssd-mobilenet.onnx","--labels=/jetson-inference/python/training/detection/ssd/models/Khas/labels.txt","--input-blob=input_0","--output-cvg=scores","--output-bbox=boxes"],threshold=0.5) I solved this issue by just strating path from /jetson-inference/...