dusty-nv/jetson-inference #413: Python example using OpenCV image

rosshemsley, 2019-09-13:
Are there any examples using Python to feed OpenCV images instead of reading files?

Context is that I have a program receiving a stream of OpenCV images (in a Ros node) and I would like to run detection.

Ideally, re-using as many resources between calls as possible.

Currently the API that only takes files means I have to round-trip to the disk, which is quite expensive.

alexabstreiter, 2020-01-03:
First, you need to convert your image to RGBA. So if your image input is and numpy array and in RGB you can do this:
input_image = cv2.cvtColor(input, cv2.COLOR_RGB2RGBA).astype(np.float32)

Next, we have to move the image to CUDA:
input_image = jetson.utils.cudaFromNumpy(input_image)

After that, the prediction works:
predictions = net.Detect(input_image, input.shape[1], input.shape[0])

Totemi1324, 2020-02-25:
And how to convert it back to Numpy-image?

alexabstreiter, 2020-02-25:
You can use `jetson.utils.cudaToNumpy` for that.

Kmarconi, 2020-03-20:
Hi, I was wondering if you had encountered any issues with the cudaToNumpy function ? When i'm using it to display my image with detections on it, I always have a white image with some blue or purple shape on it... Any guess ?

Regards :)

Totemi1324, 2020-03-20:
Make sure you convert your images into the correct color scheme and data type because CUDA handles it differently (RGBA with float32).
In my code, I want to analyze images taken by OpenCV (which are in BGR) so I need to write:
```
img = cv2.cvtColor(img, cv2.COLOR_BGR2RGB)
img = cv2.cvtColor(img, cv2.COLOR_RGB2RGBA).astype(np.float32)
img = jetson.utils.cudaFromNumpy(img)
detections = net.Detect(img, 1280, 720)
img = jetson.utils.cudaToNumpy(img, 1280, 720, 4)
img = cv2.cvtColor(img, cv2.COLOR_RGBA2RGB).astype(np.uint8)
img = cv2.cvtColor(img, cv2.COLOR_RGB2BGR)
```
Hope this helps

Kmarconi, 2020-03-23:
Thanks, worked as I wanted :)

tlhcelik, 2020-04-15:
> Make sure you convert your images into the correct color scheme and data type because CUDA handles it differently (RGBA with float32).
> In my code, I want to analyze images taken by OpenCV (which are in BGR) so I need to write:
> 
> ```
> img = cv2.cvtColor(img, cv2.COLOR_BGR2RGB)
> img = cv2.cvtColor(img, cv2.COLOR_RGB2RGBA).astype(np.float32)
> img = jetson.utils.cudaFromNumpy(img)
> detections = net.Detect(img, 1280, 720)
> img = jetson.utils.cudaToNumpy(img, 1280, 720, 4)
> img = cv2.cvtColor(img, cv2.COLOR_RGBA2RGB).astype(np.uint8)
> img = cv2.cvtColor(img, cv2.COLOR_RGB2BGR)
> ```
> 
> Not worked in my project

Totemi1324, 2020-04-15:
@tlhcelik Could you please specify what problem you are encountering? Then perhaps we can give you some help.

closed 2023-04-27 by dusty-nv