dusty-nv/jetson-inference #414: there should be a issue about the cost Walt performance of models

vxgu86, 2019-09-21:
nano always shut down suddenly due to Walt performance problem,  Even I am using a 5v4a power supply, the walt cost is somehow beyond nano's performance. 
I am wondering, did anyone else encounter the same problem?
how can I estimate the walt cost performance before deploy the model on nano?

dusty-nv, 2019-09-21:
Try running 'sudo nvpmodel -m 1' and see if that fixes the problem.  If it does, your power supply isn't delivering a consistent voltage under load.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

vxgu86, 2019-09-22:
nvpmodel -m 1 should be default mode, right?
my power supply is 
input: ac100-240v 50/60HZ
output DC5V 4A
not enough even for refined fcn-alexnet

dusty-nv, 2019-09-22:
Default mode is nvpmodel -m 0

vxgu86, 2019-09-23:
I tested and it's mode 0 ,which is 10 walt.
should it be 20=5v*4a?

vxgu86, 2019-09-26:
for  nano , mode 0 is the most powerful mode ,right? @dusty-nv
why change from 0 to 1 to test it?

dusty-nv, 2019-09-26:
If you change it to 5W mode (mode 1), and you don't get sudden shutdowns anymore, than would indicate a problem with the power supply because it can't support the full 10W mode (mode 0).

vxgu86, 2019-09-26:
thanks for your quick reply @dusty-nv .
the sudden shutdown happened in certain conditions, not just when it's running.
 for example, when I ran the imagenet-camera, nano shut down when the camera view get opened; another example is, when in detect-camera, shutdown happened when the console load the model and optimized it into the tensorrt engine.
 
let's say that , the model needs 15W to set fire, if nano could not satisfy it, it would shut down,right?
I dont know whether it's the power supply's problem or just the model need more power(W)

another question ,the 10W mode means that the cost is always 10W or that the maximum line is updated to 10W?

dusty-nv, 2019-09-26:
If it is shutting down, it's likely to be a power supply issue.  

I recommend trying one of the DC barrel jack adapters from the Jetson Nano Supported Components List like this one:  https://www.adafruit.com/product/1466

vxgu86, 2019-09-26:
there is absolutely no causes about my model consuming power problem?
eee this adapters is somehow huge for me, I am planing to put it on drone

dusty-nv, 2019-09-26:
The model isn't going to draw that much power to exceed 5V 4A (you can check the power consumption with the `tegrastats` program).

It is more likely that this power supply is unable to supply a consistent voltage.  I never have shutdown no matter the model.  Which power supply are you using?

vxgu86, 2019-10-11:
problem found to be the power supply, thanks

closed 2019-10-11 by vxgu86