dusty-nv/jetson-inference #417: jetson-utils target missing

PaulRuvu, 2019-09-27:
Hi,

After building `jetsen-inference` (as described [here](https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo.md)) I try to depend on it from a custom package. 
However the `find_package` cmake call cannot fully find it and throws the following warning:

```
CMake Warning at custom_package/CMakeLists.txt:25 (find_package):
  Found package configuration file:

    /usr/local/share/jetson-inference/cmake/jetson-inferenceConfig.cmake

  but it set jetson-inference_FOUND to FALSE so package "jetson-inference" is
  considered to be NOT FOUND.  Reason given by package:

  The following imported targets are referenced, but are missing:
  jetson-utils
```
There seems to be an installation error which prevents `jetson-utils` to be found.

Has anyone experienced anything similar before? Any help would be appreciated.

System info:
Jetson tx2
Ubuntu 16.04
ros kinetic

dusty-nv, 2019-09-27:
Did you try find_package(jetson-utils) first?

There is an example CMakeLists in the 'Creating your own Image Recognition Program (C++)' step.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

rokusottervanger, 2019-09-27:
That fixed it, thanks!