dusty-nv/jetson-inference #418: detectnet-console --network

alanmcivor1959, 2019-09-28:
If you run detectnet-console in the jetson-inference/build/aarch64/bin/ with a --network argument it works fine. If you run it in another directory it fails with errors like this

[TRT]   binding to output 1 bboxes  dims (b=1 c=4 h=28 w=28) size=12544
device GPU, /home/alan/hello_ai_world/jetson-inference/build/aarch64/bin/networks/facenet-120/snapshot_iter_24000.caffemodel initialized.
detectNet -- number object classes:   1
detectNet -- maximum bounding boxes:  784
detectNet -- loaded 1 class info entries
detectNet -- number of object classes:  1
[image] failed to find file 'facenet'
failed to load image 'facenet'

The above is on a Jetson TX2

dusty-nv, 2019-09-28:
Can you try doing 'sudo make install' and then running the program without specifying a path (i.e. just detectnet-console <args>)

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

alanmcivor1959, 2019-09-29:
I did a 'make install' and tried again without the path. The result was exactly the same.

dusty-nv, 2019-09-29:
Can you provide the command line you are using?

dslarm, 2019-10-17:
Hi - this never worked, no matter which directory you used.
CommandLine::getPosition doesn't increment i again when removing the first (--network) argument (or any argument..) - it should.

dusty-nv, 2019-10-17:
Can you try using `--network=pednet` instead of `--network pednet`?

omartin2010, 2019-12-21:
I just hit the same issue, and the same fix works. I'm not sure why it makes a difference, as I've used argparse many times and never used '=' signs with it...

closed 2023-04-27 by dusty-nv