dusty-nv/jetson-inference #421: Training coco-dog example for detectnet via digits.. no bounding box detected

jogiji, 2019-10-01:
We trained the ms-coco dog set as per the tutorial under two-days to a demo.
Training process completes successfully..but after testing with a sample image, there is no bounding box detected..similar thing happens with custom dataset.
We are using the docker image with tag 19.09-caffe

revrr, 2019-10-06:
whats the the mAP (value) on the graph? I have a similar problem and the mAP is 0. I think there is something to do with that. mAP should be above 0

jogiji, 2019-10-06:
> whats the the mAP (value) on the graph? I have a similar problem and the mAP is 0. I think there is something to do with that. mAP should be above 0

Yes the mAP stays at 0 which should ideally not happen if we follow the tutorial.. are we missing out something which is not covered for latest DIGITs release.

dipampatel18, 2020-02-04:
@jogiji @revrr Since the mAP = 0, I think the issue might be with the class names. The correct class names should be- dontcare, dog.

If you simply wrote 'dog' in the class name, that class is ignored by default in digits. You have to follow the above convention.