dusty-nv/jetson-inference #423: jetson.utils.gstCamera.captureRGBA fails on built-in TX2 camera

AGenchev, 2019-10-05:
Here there's example how to build cam recognition with python, the source:
```
import jetson.inference
import jetson.utils

net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
camera = jetson.utils.gstCamera(1280, 720, "/dev/video0")  # using V4L2
display = jetson.utils.glDisplay()

while display.IsOpen():
        img, width, height = camera.CaptureRGBA()
```
This fails on JetsonTX2 Ubuntu18 updated to the last Jetpack, etc
The problem is in the gst filters:
```
[gstreamer] gstCamera onEOS
[gstreamer] gstreamer v4l2src0 ERROR Internal data stream error.
[gstreamer] gstreamer Debugging info: gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
```

dusty-nv, 2019-10-05:
Can you try changing "/dev/video0" to "0"?

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

AGenchev, 2019-10-05:
Ya so quisk !!!! 
Yes, it works now :-) Many thanks !

closed 2019-10-05 by AGenchev

dusty-nv, 2019-10-05:
No problem - glad you got it working.  BTW, using "0" means to use the first onboard CSI camera interface instead of V4L2.

AGenchev, 2019-10-05:
Oh, now I understand - we can connect up to 6 CSI cameras there. 
The image of the built-in one is flipped though. I could get it straight with `! nvvidconv flip-method=0 ! ` converter filter in the g-streamer pipeline. If we could do this in utils.gstCamera would be nice also.

dusty-nv, 2019-10-05:
There is the flip-method set in gstCamera.cpp source file, you just need to change it there and recompile and then do sudo make install again.

AGenchev, 2019-10-05:
thanks, I saw it:
```
  if NV_TENSORRT_MAJOR > 1 && NV_TENSORRT_MAJOR < 5      // if JetPack 3.1-3.3 (different flip-method)
        const int flipMethod = 0;                       // Xavier (w/TRT5) camera is mounted inverted
  #else
        const int flipMethod = 2;
  #endif  
```
but how could you know I can deal with C++ ?!