dusty-nv/jetson-inference #427: Trouble getting width and height of bounding box

LookiePookie, 2019-10-12:
Currently im trying to call the get width and height methods of the bounding box for but its saying that method doesn't exist, this is from the detectnet-camera example.

`import jetson.inference as ji
import jetson.utils as ju

# load the object detection model
net = ji.detectNet("ssd-mobilenet-v2", threshold=0.5)

camera = ju.gstCamera(1280, 720, "0")

display = ju.glDisplay()

bbXCoord = 0
bbYCoord = 0

while display.IsOpen():
	img, width, height = camera.CaptureRGBA()
	detections = net.Detect(img, width, height)
	display.SetBackgroundColor(0.1, 0.1, 0.1, 1.0)
	display.RenderOnce(img, width, height)
	display.SetTitle("Person finder | Network {:.0f} FPS".format(net.GetNetworkFPS()))

	bbXCoord = ju.glTexture.GetWidth()
	bbYCoord = ju.glTexture.GetHeight()
	print(bbXCoord)
	print(bbYCoord)`

dusty-nv, 2019-10-14:
The bounding box coordinates come from the `detections` list:

``` python
for detection in detections:
    print(detection.Left)
    print(detection.Top)
    print(detection.Width)
    print(detection.Height)
```

LookiePookie, 2019-11-03:
Thanks so much its working now!

closed 2019-11-03 by LookiePookie