dusty-nv/jetson-inference #428: cannot successfully run jetson nano camera commands from another host with ssh -X

syonekura, 2019-10-14:
Hi, i'm able to run all the example scripts locally from the jetson nano, but now I need to access it from my pc for developing purposes. 

So I've set the nano on the same local network than my pc. The Jetson Nano is conneced to a RPi Camera v2.1, power adapter and ethernet, while the PC has the latest nvidia drivers (430.50) installed via runfile with the --no-opengl-files according to [this guide](https://gist.github.com/wangruohui/df039f0dc434d6486f5d4d098aa52d07). This was because installing via apt-get or other means got me to the error reported [here](https://stackoverflow.com/questions/40644455/x-error-of-failed-request-badvalue-integer-parameter-out-of-range-for-operatio).

But now running any script via `ssh -X` throws some errors that i dont know how to deal with. For example when running `./camera-viewer`:

```
jetson@nano:~/jetson-inference/build/aarch64/bin$ ./camera-viewer
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera attempting to initialize with GST_SOURCE_NVARGUS, camera 0
[gstreamer] gstCamera pipeline string:
nvarguscamerasrc sensor-id=0 ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, framerate=30/1, format=(string)NV12 ! nvvidconv flip-method=2 ! video/x-raw ! appsink name=mysink
nvbuf_utils: Could not get EGL display connection
[gstreamer] gstCamera successfully initialized with GST_SOURCE_NVARGUS, camera 0

camera-viewer:  successfully initialized camera device
    width:  1280
   height:  720
    depth:  12 (bpp)
[OpenGL] glDisplay -- X screen 0 resolution:  1440x900
[OpenGL] glDisplay -- display device initialized
[gstreamer] opening gstCamera for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter1
[gstreamer] gstreamer changed state from NULL to READY ==> nvvconv0
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter0
[gstreamer] gstreamer changed state from NULL to READY ==> nvarguscamerasrc0
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter1
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvvconv0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvarguscamerasrc0
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer msg new-clock ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter1
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvvconv0
[gstreamer] gstreamer msg stream-start ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvarguscamerasrc0
camera-viewer:  camera open for streaming
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3264 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3264 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 4 
   Output Stream W = 1280 H = 720 
   seconds to Run    = 0 
   Frame Rate = 120.000005 
GST_ARGUS: PowerService: requested_clock_Hz=6048000
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
[gstreamer] gstCamera onPreroll
[gstreamer] gstCamera -- allocated 16 ringbuffers, 1382400 bytes each
[gstreamer] gstreamer changed state from READY to PAUSED ==> mysink
[gstreamer] gstreamer msg async-done ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> mysink
[... 233 lines cut ...]

[cuda]  cudaGetLastError()
[cuda]     unknown error (error 30) (hex 0x1E)
[cuda]     /home/jetson/jetson-inference/utils/cuda/cudaYUV-NV12.cu:384
[cuda]  cudaNV12ToRGBA32((uint8_t*)input, (float4*)mRGBA[mLatestRGBA], mWidth, mHeight)
[cuda]     unknown error (error 30) (hex 0x1E)
[cuda]     /home/jetson/jetson-inference/utils/camera/gstCamera.cpp:269
[gstreamer] gstCamera failed to convert frame to RGBA
Traceback (most recent call last):
  File "./imagenet-camera.py", line 57, in <module>
    img, width, height = camera.CaptureRGBA()
Exception: jetson.utils -- gstCamera failed to CaptureRGBA()
jetson.utils -- PyDisplay_Dealloc()
jetson.utils -- PyCamera_Dealloc()
[gstreamer] closing gstCamera for streaming, transitioning pipeline to GST_STATE_NULL
GST_ARGUS: Cleaning up
CONSUMER: Done Success
GST_ARGUS: Done Success
PyTensorNet_Dealloc()
jetson.utils -- PyFont_Dealloc()
GST_ARGUS: 
PowerServiceHwVic::cleanupResources
```

I'm not familiar either with cuda or opengl, could someone please help me?

dusty-nv, 2019-10-14:
Hi @syonekura , the OpenGL rendering with CUDA interopability that this project uses is meant to run on a display directly connected to the Nano, and is unlikely to work over SSH forwarding.  And even if it were, it would display very slowly, so it is recommended to view the camera video from the Nano itself.

aansrajpoot, 2021-07-03:
Hi @syonekura , I'm just simple trying to access my Jetson Nano camera via SSH connection. But I'm getting an error of this. I'm not able to run my python file to launch camera. I have to detect lane lines in live video streaming.
Thank you  if you can help me.

[image]

benb0jangles, 2021-09-15:
> Hi @syonekura , the OpenGL rendering with CUDA interopability that this project uses is meant to run on a display directly connected to the Nano, and is unlikely to work over SSH forwarding. And [...]

Hey, would it be possible to use ssh to run jetson-inference examples yet have the camera image show on the display connected to jetson-nano hdmi?

dusty-nv, 2021-09-15:
Hmm, maybe if you set the DISPLAY environment variable appropriately.  DISPLAY=:0
Not sure though, haven't tried it.

closed 2023-04-27 by dusty-nv