dusty-nv/jetson-inference #430: How to use YOLOV3 caffe in deep stream?

MuhammadAsadJaved, 2019-10-16:
@all 
How can I use YOLOV3 caffe pre-trained model in deep stream? 

I have yolov3.prototxt and yolov3.caffemodel (weights).

dusty-nv, 2019-10-16:
See here for guide for using custom YOLO model in DeepStream 4.0:   https://docs.nvidia.com/metropolis/deepstream/4.0/Custom_YOLO_Model_in_the_DeepStream_YOLO_App.pdf

For further assistance with DeepStream, please search or post to the DeepStream SDK forum:  https://devtalk.nvidia.com/default/board/209/deepstream-sdk/

closed 2019-10-16 by dusty-nv

MuhammadAsadJaved, 2019-10-29:
@dusty-nv  Yes I understand how to use YOLOv3 caffe implementation in deepstream.  can you tell me how to use Tensorflow models trained using TensorFlow object detection API?  I have trained Faster_rcnn_inception_v2 and  ssd_resnet50_v1 for boat detection using the TensorFlow object detection API. Now I have checkpoint,  frozen_interface_graph.pb , model.ckpt.data-00000-of-00001 , model.ckpt.index , model.ckpt.meta  pipeline.config files for both models.

The second i have trained keras_yolov2 models as well. you can explain any method either tensorlfow or keras_yolov2.   I have searched and tried a lot but failed to run.

MuhammadAsadJaved, 2020-06-05:
> See here for guide for using custom YOLO model in DeepStream 4.0: https://docs.nvidia.com/metropolis/deepstream/4.0/Custom_YOLO_Model_in_the_DeepStream_YOLO_App.pdf
>
> For further assistance with [...]

Sir these are details about Darknet yolov3 model. Not about Caffe YOLOV3. Is there any documentation about Caffe YOLOV3?