dusty-nv/jetson-inference #432: detectnet-console.py failed

ccl-private, 2019-10-18:
when I run "./detectnet-console.py --network=ssd-mobilenet-v2 images/peds_0.jpg output.jpg "
It output this:
###########################################################################
jetson.inference.__init__.py
jetson.inference -- initializing Python 2.7 bindings...
jetson.inference -- registering module types...
jetson.inference -- done registering module types
jetson.inference -- done Python 2.7 binding initialization
jetson.utils.__init__.py
jetson.utils -- initializing Python 2.7 bindings...
jetson.utils -- registering module functions...
jetson.utils -- done registering module functions
jetson.utils -- registering module types...
jetson.utils -- done registering module types
jetson.utils -- done Python 2.7 binding initialization
[image] loaded 'images/peds_0.jpg'  (1920 x 1080, 3 channels)
jetson.inference -- PyTensorNet_New()
jetson.inference -- PyDetectNet_Init()
jetson.inference -- detectNet loading network using argv command line params
jetson.inference -- detectNet.__init__() argv[0] = './detectnet-console.py'
jetson.inference -- detectNet.__init__() argv[1] = '--network=ssd-mobilenet-v2'
jetson.inference -- detectNet.__init__() argv[2] = 'images/peds_0.jpg'
jetson.inference -- detectNet.__init__() argv[3] = 'output.jpg'

detectNet -- loading detection network model from:
          -- prototxt     NULL
          -- model        ssd-mobilenet-v2
          -- input_blob   'data'
          -- output_cvg   'coverage'
          -- output_bbox  'bboxes'
          -- mean_pixel   0.000000
          -- mean_binary  NULL
          -- class_labels NULL
          -- threshold    0.500000
          -- batch_size   1

[TRT]   TensorRT version 4.0.2
[TRT]   detected model format - custom  (extension '.ssd-mobilenet-v2')
[TRT]   model format 'custom' not supported by jetson-inference
detectNet -- failed to initialize.
jetson.inference -- detectNet failed to load built-in network 'ssd-mobilenet-v2'
PyTensorNet_Dealloc()
Traceback (most recent call last):
  File "./detectnet-console.py", line 51, in <module>
    net = jetson.inference.detectNet(opt.network, sys.argv, opt.threshold)
Exception: jetson.inference -- detectNet failed to load network
jetson.utils -- freeing CUDA mapped memory
##############################################################################

I just follow the guide, I don't know why this issue occurred.

dusty-nv, 2019-10-18:
The SSD-Mobilenet and SSD-Inception models require TensorRT 5.0 or newer.  I see from the log you are using TensorRT 4.0.  Which version of JetPack are you using?

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

ccl-private, 2019-10-18:
my JetPack is L4T-3.3. Can I upgrade TensorRT only, or I need to re-flash my TX2 using newer JetPack?

dusty-nv, 2019-10-18:
You would need to reflash to upgrade.

ccl-private, 2019-10-18:
Thank you for reply～

closed 2019-10-18 by ccl-private

WhaSukGO, 2020-05-05:
```
[TRT]   TensorRT version 6.0.1
[TRT]   loading NVIDIA plugins...
[TRT]   Plugin Creator registration succeeded - GridAnchor_TRT
[TRT]   Plugin Creator registration succeeded - GridAnchorRect_TRT
[TRT]   Plugin Creator registration succeeded - NMS_TRT
[TRT]   Plugin Creator registration succeeded - Reorg_TRT
[TRT]   Plugin Creator registration succeeded - Region_TRT
[TRT]   Plugin Creator registration succeeded - Clip_TRT
[TRT]   Plugin Creator registration succeeded - LReLU_TRT
[TRT]   Plugin Creator registration succeeded - PriorBox_TRT
[TRT]   Plugin Creator registration succeeded - Normalize_TRT
[TRT]   Plugin Creator registration succeeded - RPROI_TRT
[TRT]   Plugin Creator registration succeeded - BatchedNMS_TRT
[TRT]   Could not register plugin creator:  FlattenConcat_TRT in namespace: 
[TRT]   completed loading NVIDIA plugins.
[TRT]   detected model format - custom  (extension '.pb')
[TRT]   model format 'custom' not supported by jetson-inference
detectNet -- failed to initialize.
jetson.inference -- detectNet failed to load built-in network 'ssd-mobilenet-v2'
PyTensorNet_Dealloc()
Traceback (most recent call last):
  File "./detectnet-camera.py", line 49, in <module>
    net = jetson.inference.detectNet(opt.network, sys.argv, opt.threshold)
Exception: jetson.inference -- detectNet failed to load network

```

I'm having the same issue. My Jetpack version is 4.3 and verified that TensorRT is 6.0.

The model I'm using is ```ssd_resnet_50_fpn_coco``` from [here](https://github.com/tensorflow/models/blob/master/research/object_detection/g3doc/detection_model_zoo.md)