dusty-nv/jetson-inference #433: How to use Pytorch-cat-dog tutorial with object detection Onnx model instead?

Tylersuard, 2019-10-18:
Hello!  I really enjoy your tutorials, especially the one about re-training a neural network.  I had no idea it was that easy!

I have trained my own model using Microsoft's Custom Vision suite, and exported it as Tensorflow, Tensorflow Lite, and Onnx.  This is an object detection model.

I was wondering how to load my model the same way you did in the pytorch-cat-dog tutorial, but using the model mentioned above instead?  I'd like that same super-fast speed.

Thank you!

dusty-nv, 2019-10-19:
Hi Tyler, glad you enjoyed it!

The first thing to test, would be if TensorRT can load your ONNX model.  You can easily test that with the trt-exec tool, found in /usr/src/tensorrt/bin

If it loads, you would need to modify detectNet.cpp to perform the needed pre-processing (generally applying mean subtraction, std deviation, and CHW format - the same pre-processing as done during training / validation). Also will need the post-processing code for interpreting the output layers.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

Tylersuard, 2019-10-19:
Ok great!  I will test the model as suggested.  However, I created the model using Azure Custom Vision, which does everything automatically and I have no idea what preprocessing or post-processing was applied to the model. Any recommendations for me there?

I’m basically just hoping to use my exported model on the Nano with 20fps inference speed, like in the demos. Thank you!

closed 2023-04-27 by dusty-nv