dusty-nv/jetson-inference #434: camera.CaptureRGBA() hangs

BouweCeunen, 2019-10-19:
The `img, width, height = camera.CaptureRGBA()` function just hangs after approximately 30minutes of running. I can't find the source code where this function returns the img, width and height. Anyone any idea? Thanks.

ledaiduongvnth, 2020-09-17:
I have the same issue. Please help!

ledaiduongvnth, 2020-09-17:
It seems the timeout does not work

dusty-nv, 2020-09-17:
Can you try using the videoSource interface?   see here for example:  

* https://github.com/dusty-nv/jetson-utils/blob/916023814a1ac1401380c5d4313454c78c0a5f40/python/examples/video-viewer.py
* https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md

Also, what happens if you try a timeout like `timeout=100`?

closed 2023-04-27 by dusty-nv