dusty-nv/jetson-inference #436: detectnet_camera what(): CHECK failed: (index) < (current_size_): Aborted (core dumped)

pacejohn, 2019-10-30:
Hello,

I have trained a caffe object detection model using a different software (IBM Power AI Vision).  When I try to run the following:

./detectnet-camera --model /home/pace/jetson-inference/data/networks/PAIV/23ba8fbe-b645-42cc-b551-e16d98e5debc.caffemodel --prototxt /home/pace/jetson-inference/data/networks/PAIV/test.prototxt --class_labels /home/pace/jetson-inference/data/networks/PAIV/classname.txt --camera=/dev/video0

I get the following error:

[TRT]   attempting to open engine cache file /home/pace/jetson-inference/data/networks/PAIV/23ba8fbe-b645-42cc-b551-e16d98e5debc.caffemodel.1.1.GPU.FP16.engine
[TRT]   cache file not found, profiling network model on device GPU
[TRT]   device GPU, loading /home/pace/jetson-inference/data/networks/PAIV/test.prototxt /home/pace/jetson-inference/data/networks/PAIV/23ba8fbe-b645-42cc-b551-e16d98e5debc.caffemodel
[libprotobuf FATAL /home/erisuser/p4sw/sw/gpgpu/MachineLearning/DIT/externals/protobuf/aarch64/10.0/include/google/protobuf/repeated_field.h:1078] CHECK failed: (index) < (current_size_):
terminate called after throwing an instance of 'google_private::protobuf::FatalException'
  what():  CHECK failed: (index) < (current_size_):
Aborted (core dumped)

I never creates the .engine file like it does with the other models the first time I run them.  For example when I first ran DetectNet-COCO-Dog, it created snapshot_iter_38600.caffemodel.1.1.GPU.FP16.engine.  So I'm guessing the error occurs when converting the model into TensorRT format, so there is something wrong with the caffe model that PAIV exported.  I do not know how to find the information, though, since the error is pretty vague.  I'm not sure where error log files are located or I would look at them.  

I can paste in any of the files you may need or put them in Dropbox if that would help. 

Thanks in advance.

dusty-nv, 2019-10-30:
You could try running the model with trt-exec first to test that TensorRT is able to load the model.  It can be found under /usr/src/tensorrt/bin

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

pacejohn, 2019-10-30:
Same error using trt-exec.  I'll keep investigating on my end.  Let me know if you think of anything else.

closed 2023-04-27 by dusty-nv