dusty-nv/jetson-inference #440: something  fault running imagenet-console on jetson nano

tonyprince6, 2019-11-02:
i do not how to this problem.googlenet has been downloaded.
~/jetson-inference/build/aarch64/bin$ ./imagenet-console --network=googlenet images/orange_0.jpg output_0.jpg

imageNet -- loading classification network model from:
         -- prototxt     networks/googlenet.prototxt
         -- model        networks/bvlc_googlenet.caffemodel
         -- class_labels networks/ilsvrc12_synset_words.txt
         -- input_blob   'data'
         -- output_blob  'prob'
         -- batch_size   1

[TRT]   TensorRT version 5.0.6
[TRT]   loading NVIDIA plugins...
[TRT]   completed loading NVIDIA plugins.
[TRT]   detected model format - caffe  (extension '.caffemodel')
[TRT]   desired precision specified for GPU: FASTEST
[TRT]   requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]   native precisions detected for GPU:  FP32, FP16
[TRT]   selecting fastest native precision for GPU:  FP16
[TRT]   attempting to open engine cache file networks/bvlc_googlenet.caffemodel.1.1.GPU.FP16.engine
[TRT]   cache file not found, profiling network model on device GPU
[TRT]   device GPU, loading  networks/bvlc_googlenet.caffemodel
[TRT]   CaffeParser: Could not open file 
[TRT]   CaffeParser: Could not parse deploy file
[TRT]   device GPU, failed to parse caffe network
[TRT]   device GPU, failed to load networks/bvlc_googlenet.caffemodel
[TRT]   failed to load networks/bvlc_googlenet.caffemodel
[TRT]   imageNet -- failed to initialize.
imagenet-console:   failed to initialize imageNet

dusty-nv, 2019-11-03:
Try re-downloading the models with the model downloader script.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

tonyprince6, 2019-11-04:
thank you for your reply.  I found the reason.  This is because I gave the wrong name to googlenet. prototxt

closed 2023-04-27 by dusty-nv