dusty-nv/jetson-inference #454: Semantic Segmentation models inference performance compare between pytorch & tensorrt

hey1996, 2019-11-27:
I'm trying to test my Semantic Segmentation custom model's  performance on nano, such as Fps, accuracy, IoU, etc. Then I saw there are some pre-trained models' performance test results on README.md. It looks like what i'm searching for! Could you provide the test code?? Thanks~

dusty-nv, 2019-11-27:
Hi @hey1996 , here is the PyTorch training code use for the FCN-ResNet18 models:   https://github.com/dusty-nv/pytorch-segmentation

These scripts include measuring the IoU and accuracy during training/eval, which is where the numbers in the table came from.  The FPS numbers were gathered onboard the Nano using the `benchmark_models.sh` tool:   https://github.com/dusty-nv/jetson-inference/blob/master/tools/benchmark-models.sh

hey1996, 2019-12-01:
Hi @dusty-nv, thank for your helping!!

closed 2019-12-01 by hey1996