dusty-nv/jetson-inference #460: sudo make install get error : file INSTALL cannot find!!

brucelee78, 2019-12-03:
hello
i compiling the repo according to
[this markdown file](
https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo.md#compiling-the-project)

but when i was going to :

`sudo make install`

i get a error, and i can't solve it!

```
-- Up-to-date: /usr/lib/python3.6/dist-packages/Jetson
-- Up-to-date: /usr/lib/python3.6/dist-packages/Jetson/Inference
-- Up-to-date: /usr/lib/python3.6/dist-packages/Jetson/Inference/__init__.py
-- Installing: /usr/lib/python3.6/dist-packages/Jetson/__init__.py
CMake Error at python/bindings_python_3.6/cmake_install.cmake:89 (file):
  file INSTALL cannot find
  "/home/nvidia/jetson-inference/python/bindings/../python/jetson".
Call Stack (most recent call first):
  python/cmake_install.cmake:78 (include)
  cmake_install.cmake:129 (include)
```

please help me ,thank you!
my TX2 is jetpack 4.2.1!

dusty-nv, 2019-12-03:
Hi @brucelee78, I haven't seen this error before and am not sure what the problem is - maybe these got deleted somehow?  You could try re-cloning the repo, or you could try commenting out these two lines (56-57):  https://github.com/dusty-nv/jetson-inference/blob/87b5a8814a60e860709b35f3f774907c249db081/python/bindings/CMakeLists.txt#L56

Then re-run `cmake ../` , `make`, and `sudo make install`

Then copy the two folders found under `jetson-inference/python/python` to `/usr/lib/python3.6/dist-packages`

brucelee78, 2019-12-04:
@dusty-nv，thanks for your reply! i have solved this problem!
according to your advice，and i make some other changes.
i rename `jetson-inference/python/python/Jetson` to `jetson`
and `jetson-inference/utils/python/python/Jetson` to `jetson`

in addition,comment out `jetson-inference/python/bindings/CmakeList.txt` line 56
and  `jetson-inference/utils/python/bindings//CmakeList.txt` line 76

and Then re-run cmake ../ , make, and sudo make install

problem is solved!!
thank for your advice again!

csjryt, 2022-05-01:
> @dusty-nv，thanks for your reply! i have solved this problem! according to your advice，and i make some other changes. i rename `jetson-inference/python/python/Jetson` to `jetson` and [...]

Thanks, that help me out :)

y2934, 2023-02-27:
why am I still wrong???

y2934, 2023-02-27:
thank you very much but are line 56 and line 76 annotated all the time?

dusty-nv, 2023-02-27:
Are you using an NTFS-formatted drive or partition perhaps?   It was found that NTFS is case insensitive.  But ext3/ext4 (linux) works correctly.  You may run into other issues with NTFS too, like with the symbolic links

closed 2023-04-27 by dusty-nv

Buckspitz, 2024-10-23:
there are two files needed to be edit.  jetson-inference/python/bindings/CmakeList.txt and  jetson-inference/utils/python/bindings/CmakeList.txt . The second to last line of the document is the same as the third to last line. Since the files' name use 'Jetson' rather than 'jetson' so you need to delete the penultimate line.

Buckspitz, 2024-10-23:
jetson-inference/python/bindings/CmakeList.txt and jetson-inference/utils/python/bindings/CmakeList.txt这两个文件的倒数第二行和倒数第三行是同样的代码。
 install(DIRECTORY ../python/Jetson DESTINATION ${PYTHON_BINDING_INSTALL_DIR})
 install(DIRECTORY ../python/Jetson DESTINATION ${PYTHON_BINDING_INSTALL_DIR})
 由于文件加名称是Jetson，首字母是大写的，所以需要删除或者注释掉路径是小写j开头的jetson的这行代码