dusty-nv/jetson-inference #461: v4l2src0 error internal data stream error

kenken64, 2019-12-04:
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera attempting to initialize with GST_SOURCE_NVARGUS, camera /dev/video0
[gstreamer] gstCamera pipeline string:
v4l2src device=/dev/video0 ! video/x-raw, width=(int)1920, height=(int)720, format=YUY2 ! videoconvert ! video/x-raw, format=RGB ! videoconvert !appsink name=mysink
[gstreamer] gstCamera successfully initialized with GST_SOURCE_V4L2, camera /dev/video0
jetson.utils -- PyDisplay_New()
jetson.utils -- PyDisplay_Init()
[OpenGL] glDisplay -- X screen 0 resolution:  1920x1080
[OpenGL] glDisplay -- display device initialized
100mm

[gstreamer] opening gstCamera for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> videoconvert1
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter1
[gstreamer] gstreamer changed state from NULL to READY ==> videoconvert0
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter0
[gstreamer] gstreamer changed state from NULL to READY ==> v4l2src0
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline0
[gstreamer] gstreamer changed state from READY to PAUSED ==> videoconvert1
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter1
[gstreamer] gstreamer changed state from READY to PAUSED ==> videoconvert0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> v4l2src0
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline0
[gstreamer] gstreamer msg new-clock ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> videoconvert1
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter1
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> videoconvert0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> v4l2src0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer msg stream-start ==> pipeline0
[gstreamer] gstCamera onEOS
[gstreamer] gstreamer v4l2src0 ERROR Internal data stream error.
[gstreamer] gstreamer Debugging info: gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
[gstreamer] gstreamer changed state from READY to PAUSED ==> mysink

Matesanz, 2020-01-25:
Did you solve it?

ajschleg, 2020-02-15:
I am facing the same issue, where you able to solve?

gasparramoa, 2020-02-17:
Hi!!! Probably is because you don't have the correct width and height specified for the device which is normally 640 * 480

By default, the discrete size is 1920 * 720.
> v4l2src device=/dev/video0 ! video/x-raw, width=(int)1920, height=(int)720, format=YUY2 !

Just specify the discrete size like this: 
`python3 imagenet-camera.py --camera=/dev/video0 --width=640 --heigh=480`
or
`python3 detectnet-camera.py --camera=/dev/video0 --width=640 --heigh=480`

WSPluta, 2020-02-17:
> Hi!!! Probably is because you don't have the correct width and height specified for the device which is normally 640 * 480
>
> By default, the discrete size is 1920 * 720.
>
>
> Just specify the discrete [...]

not sure what you are referring ? 
same problem at my end,

where would that be if source code is 
`import jetson.inference
import jetson.utils

net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
camera = jetson.utils.gstCamera(1280, 720, "/dev/video0")  # using V4L2
display = jetson.utils.glDisplay()

while display.IsOpen():
	img, width, height = camera.CaptureRGBA()
	detections = net.Detect(img, width, height)
	display.RenderOnce(img, width, height)
	display.SetTitle("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))`

WSPluta, 2020-02-17:
I got it, in my case Im using raspberry pi v2.1 mcp cam.

Therefore I needed to change 
`camera = jetson.utils.gstCamera(1280, 720, "/dev/video0") # using V4L2`

To

`camera = jetson.utils.gstCamera(1280, 720, "0") # using V4L2`

Give it a go!

flurpo, 2020-05-06:
I have the same exact response running ./detectnet-camera with camera=/dev/video1 using v4l2, I've tried changing resolution settings, models, ect....no luck! Any ideas? Best.

flurpo, 2020-05-12:
I found a program in the /aarch64/bin directory labeled 'v4l2-display'. This program appears to analyze and output the stream at /dev/videoX. I matched the resolution settings but still got the same response (above at output). I don't think the resolution is causing this....Best.

CouchMango, 2020-05-20:
> I got it, in my case Im using raspberry pi v2.1 mcp cam.
> 
> Therefore I needed to change
> `camera = jetson.utils.gstCamera(1280, 720, "/dev/video0") # using V4L2`
> 
> To
> 
> `camera = jetson.utils.gstCamera(1280, 720, "0") # using V4L2`
> 
> Give it a go!

This worked for me! 

I was using a USB camera and the ribbon cable camera simultaneously to see which would work. After changing the "/dev/video0" to "0", the ribbon cable camera started working upon executing the script!

sohaibma, 2020-06-01:
I am getting exactly the same error with my USB cameras. I have virtually tried every fix, so to speak, with 2 different USB cams, the problem persists. Here's what I have cross-checked:
-- camera specs including **_resolution_** & input channel using `v4l2-ctl --list-devices` & `v4l2-ctl --list-formats-ext -d /dev/video0` (since my cam is at video0), and entering appropriate values in `.gstCamera`

I have devoured multiple threads on other forums regarding this issue, and from I gather, it might have something to do with image conversion from _raw_ to RGB format, or simply some missing attributes that are stalling the _pipeline_.

Any other ideas?

sohaibma, 2020-06-01:
In addition to my previous post, can anyone also guide as to how to change the _pixel_ format? Default is set to YUY2, while both my cams show JPEG as the only supported pixel format. 

The sample code below that I'm using makes use of gstreamer APIs, and I cant seem to figure how to set or change the default pixel format setting. `.gstCamera` only has 3 attributes, and none of that is for pixel format.

`import jetson.inference
import jetson.utils

net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
camera = jetson.utils.gstCamera(640, 480, "/dev/video0")
display = jetson.utils.glDisplay()

while display.IsOpen():
	img, width, height = camera.CaptureRGBA()
	detections = net.Detect(img, width, height)
	display.RenderOnce(img, width, height)
	display.SetTitle("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))`

dusty-nv, 2020-06-01:
You would need to change the gstreamer pipeline inside of gstCamera.cpp to support jpeg decode, but I haven't tried this.  Alternatively you could see if openCV cv2 capture already supports it.

Mohit-Ak, 2020-06-17:
> You would need to change the gstreamer pipeline inside of gstCamera.cpp to support jpeg decode, but I haven't tried this. Alternatively you could see if openCV cv2 capture already supports it.

Is there a sample to feed in the image from open-cv instead? The following line does not accept an opencv image even after converting to RGBA
```detections = net.Detect(img, width, height)```

OpenCV Code
```
       image=cv2.cvtColor (frame, cv2.COLOR_RGB2RGBA)
       h, w, c = image.shape
       return image, w, h

```

dusty-nv, 2020-06-17:
@Mohit-Ak, see the jetson.utils.cudaFromNumpy() function and the cuda-from-numpy.py example from jetson-utils.  cv2 images are actually numpy arrays.

Also, if you check out my `dev` branch, I have added support for jpeg decode to gstCamera (and h264/h265/vp8/vp9 decode).  You can also play video IP streams (RTP/RTSP) or files, and encode to file or RTP IP stream.

Mohit-Ak, 2020-06-17:
> @Mohit-Ak, see the jetson.utils.cudaFromNumpy() function and the cuda-from-numpy.py example from jetson-utils. cv2 images are actually numpy arrays.
>
> Also, if you check out my `dev` branch, I have [...]

Thank you for the quick response.
I copied the camera folder from the ```dev``` branch and I saw you "[mjpeg](https://github.com/dusty-nv/jetson-utils/blob/b987c406749e140a8532594dcc4540b12b6b9ea6/camera/gstCamera.cpp)" implementation but I am not sure how to pass it as an input.

For example, if my mjpeg feed URL is "192.168.0.107/video2.mjpg" how do I pass it to gstCamera function.
Is it like this?
```camera = jetson.utils.gstCamera(960, 720, "http://192.168.0.107/video2.mjpg")```

dusty-nv, 2020-06-17:
> I copied the camera folder from the `dev` branch and I saw you "[mjpeg](https://github.com/dusty-nv/jetson-utils/blob/b987c406749e140a8532594dcc4540b12b6b9ea6/camera/gstCamera.cpp)" implementation but I am not sure how to pass it as an input.

You can't copy just the camera folder from the dev branch, you need to clone the whole dev branch.  There have been a lot of other changes to enable it - what you would now use is called the `videoSource` API (it is very similar, but more powerful underneath).

See here for an example on using it:   https://github.com/dusty-nv/jetson-utils/blob/master/python/examples/video-viewer.py

Is that URL an RTSP stream?   If so, you could try:

`video-viewer.py rtsp://192.168.0.107/video2.mjpg`

Other protocols include `rtp://` and `file://` for local video files.   
For attached cameras, there is `csi://0` and `v4l2:///dev/video0`   (replace `0` with camera number)