dusty-nv/jetson-inference #462: How to setup workflow (Jetson Nano + Ubuntu PC)

r8d8, 2019-12-06:
Hi, 
Can someone explaine me what is a proper way to setup workflow for Jetson app development?
For a past 2 days I can't compile `jetson-inference` for a x86 Ubuntu machin, each time there is some conflicts or missing libs/headers/whatever stuff.
There is totally no docs how one can develope app on host PC. Jetpack is only for Jetsons, Nvidia SDK Manager lacks TensorRT, manual installation... don't want even talk about it.

Jetson family are ARM64, so there is no IDE.
Any help will be really apreciated. Realy!

dusty-nv, 2019-12-06:
Hi @r8d8, I haven't tried to cross-compile `jetson-inference`, it is intended to be compiled natively on the Jetson.  Sorry for the trouble.

r8d8, 2019-12-06:
@dusty-nv , it's ok. I appreciate your work.
I am just trying to figure out how to develope for Jetson.
Like all Jetson devices are ARM based, so I bet no one develope directly on them (not much toling for platform). How do you developed this lib? 
I need small demo to detect persons, crop image of them from a frame and send it to server.

dusty-nv, 2019-12-06:
I personally use GEdit / terminal to develop, however if you search around the Jetson Devtalk forums, there are folks also using Visual Studio Code along with some other IDE's.  There are also some folks who cross-compile their projects, particularly the Linux kernel.

r8d8, 2019-12-06:
And how about debuging?

dusty-nv, 2019-12-06:
It would appear that VS Code can use GDB, others like code::blocks probably can too.

See these threads from the forums:

https://devtalk.nvidia.com/default/topic/1056104/jetson-nano/programing-jetson-nano-with-c-/post/5356971/#5356971

https://devtalk.nvidia.com/default/topic/1050570/jetson-nano/cross-compiling-c-c-code-for-jetson-nano-in-ubuntu-18-04-in-a-host-pc-x86-x64-/post/5331806/#5331806

You could try Nsight Eclipse Edition as a remote IDE.

closed 2023-04-27 by dusty-nv