dusty-nv/jetson-inference #466: Python 3.7 wasn't found

lbq779660843, 2019-12-15:
when I run the command `'cmake ../ ',` it  throw the error `-- Python 3.7 wasn't found
-- Copying /home/lbq/jetson-inference/utils/python/examples/camera-viewer.py
-- Copying /home/lbq/jetson-inference/utils/python/examples/cuda-from-numpy.py
-- Copying /home/lbq/jetson-inference/utils/python/examples/cuda-to-numpy.py
-- Copying /home/lbq/jetson-inference/utils/python/examples/gl-display-test.py
-- trying to build Python bindings for Python versions:  2.7;3.6;3.7
-- detecting Python 2.7...
-- found Python version:  2.7 (2.7.15+)
-- found Python include:  /usr/include/python2.7
-- found Python library:  /usr/lib/aarch64-linux-gnu/libpython2.7.so
-- detecting Python 3.6...
-- found Python version:  3.6 (3.6.9)
-- found Python include:  /usr/include/python3.6m
-- found Python library:  /usr/lib/aarch64-linux-gnu/libpython3.6m.so
-- detecting Python 3.7...
-- Python 3.7 wasn't found
-- Copying /home/lbq/jetson-inference/python/examples/detectnet-camera.py
-- Copying /home/lbq/jetson-inference/python/examples/detectnet-console.py
-- Copying /home/lbq/jetson-inference/python/examples/imagenet-camera.py
-- Copying /home/lbq/jetson-inference/python/examples/imagenet-console.py
-- Copying /home/lbq/jetson-inference/python/examples/my-detection.py
-- Copying /home/lbq/jetson-inference/python/examples/my-recognition.py
-- Copying /home/lbq/jetson-inference/python/examples/segnet-batch.py
-- Copying /home/lbq/jetson-inference/python/examples/segnet-camera.py
-- Copying /home/lbq/jetson-inference/python/examples/segnet-console.py
-- Configuring incomplete, errors occurred!
See also "/home/lbq/jetson-inference/build/CMakeFiles/CMakeOutput.log".
See also "/home/lbq/jetson-inference/build/CMakeFiles/CMakeError.log".
`
How can I fix it?thanx.

dusty-nv, 2019-12-15:
It found Python 2.7 and 3.6 on your system, 3.7 probably wasn't found because you dont have it installed.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

lbq779660843, 2019-12-16:
Hi Dustin,
I refer [this](https://github.com/dusty-nv/jetson-inference/issues/360) and  use the command:
`sudo apt-get install libpython3.7-dev`
then I check the libpython3.7 and libpython3.7m.so and they are truly installed in the same location of libpython3.6 libpython3.6m.so, however it still throw the same message.
ps: I installed Archiconda3 in my jetson nano but its base enviroment is python3.7 and I can't make a new python3.6 env because it will throw a error so I cann't install the pytorch you have shared in the jetson nano's forum.

jamesthesken, 2020-02-11:
@lbq779660843 have you solved this issue?

lbq779660843, 2020-02-25:
> @lbq779660843 have you solved this issue?

No, I'm using the yolov3 now.
Frankly speaking, the official tutorial is hard to use.

lbq779660843, 2020-03-12:
> @lbq779660843 have you solved this issue?
Still not. But I can run this tutorial now. It is weird.

closed 2020-03-12 by lbq779660843