dusty-nv/jetson-inference #467: cudaAllocMapped failed when imgCPU needs larger size

thunder95, 2019-12-19:
I am trying a video classification problem (stnet), the input size is 5268480 which is supposed to be larger than the stack limitation (float* imgCPU).

It works with the mapping process between imgCPU and imgCUDA quite well. However it threw error when running the inference with regarding of convolution. Possibly that is because I allocated the large size memory on the heap like this way:  float* imgCPU = new float[5268480];

Moreover,  I even cannot assign more values to imgCPU when use float* imgCPU=NULL, otherwise it comes to a segmentation fault. If I set the input size smaller, it works quite well also.

What should I do except for cudaMemcpy? If I copy memory from host to device, the inference would consume much more time. Thanks for your help.

dusty-nv, 2019-12-19:
Hi @thunder95, it's unclear how you are calling `cudaAllocMapped()`, can you share your code snippet?   Allocating 5MB shouldn't be a problem (or 20MB, it's unclear if the input size is in bytes or number of floats) , unless you are already out of memory.   

You can't run the inference with memory allocated like this:  `float* imgCPU = new float[5268480];` 
You need to pass a GPU pointer to the inference, or a pointer allocated with `cudaAllocMapped()`

thunder95, 2019-12-20:
> Hi @thunder95, it's unclear how you are calling `cudaAllocMapped()`, can you share your code snippet? Allocating 5MB shouldn't be a problem (or 20MB, it's unclear if the input size is in bytes or [...]

Hello, @dusty-nv , many thanks for your reply firstly.  From the hints of your reply, I checked my code again, and found I forgot the sizeof(float), and now it works. thanks again.

5268480 indicates the number of floats. The code snippet of calling `cudaAllocMapped()` as follows:

```
int INPUT_DIM = 5268480; //the input dimension: 7*15*224*224
float* imgCPU = NULL;
float* imgCUDA   = NULL;
if(!cudaAllocMapped((void**)&imgCPU, (void**)&imgCUDA, INPUT_DIM * sizeof(float) )) {
    return false;
}
for (int i = 0; i < INPUT_DIM; i++) {
     imgCPU[i] = 1.0;
}
```

closed 2019-12-20 by thunder95