dusty-nv/jetson-inference #468: Unable to performance Inference on Numpy arrays (cv2)

mbufi, 2019-12-19:
Hello,
Currently, we capture images as such:
`img, width, height = camera.CaptureRGBA()`

I would like to capture images with cv2 instead and then do inference on the numpy arrays. How can one do this? Something like this is what I invision but cannot get it to work. 

```
cap = cv2.VideoCapture(0)

while(True):
   # Capture frame-by-frame
   ret, frame = cap.read()
   height, width, channels = frame.shape

   # classify the image
   class_idx, confidence = net.Classify(frame, width, height)

```

mbufi, 2020-01-02:
Anyone else run into this ?

alexabstreiter, 2020-01-03:
First, you need to convert your image to RGBA. So if your image `input` is and numpy array and in RGB you can do this:
`input_image = cv2.cvtColor(input, cv2.COLOR_RGB2RGBA).astype(np.float32)`

Next, we have to move the image to CUDA:
`input_image = jetson.utils.cudaFromNumpy(input_image)`

After that, the prediction works:
`predictions = net.Detect(input_image, input.shape[1], input.shape[0])`

mbufi, 2020-01-03:
Thank you for the comment @alexabstreiter 
Quick question: 
When I load that image with cudaFromNumpy, do I have to do some deallocation after classification is done?

Thanks!

closed 2020-01-06 by mbufi