dusty-nv/jetson-inference #474: Any way I can disable logging?

magnusja, 2020-01-09:
Hey there,

during inference stdout is flooded with thouds of messages like this:

```
jetson.utils -- cudaFromNumpy()  ndarray dim 0 = 480                                     
jetson.utils -- cudaFromNumpy()  ndarray dim 1 = 640
jetson.utils -- cudaFromNumpy()  ndarray dim 2 = 4
jetson.utils -- freeing CUDA mapped memory                                         
jetson.inference -- PyDetection_Dealloc()
jetson.inference -- PyDetection_Dealloc()                        
jetson.utils -- cudaFromNumpy()  ndarray dim 0 = 480
jetson.utils -- cudaFromNumpy()  ndarray dim 1 = 640        
jetson.utils -- cudaFromNumpy()  ndarray dim 2 = 4  
jetson.utils -- freeing CUDA mapped memory                                   
jetson.inference -- PyDetection_Dealloc() 
jetson.inference -- PyDetection_Dealloc()
jetson.inference -- PyDetection_Dealloc()
jetson.utils -- cudaFromNumpy()  ndarray dim 0 = 480
jetson.utils -- cudaFromNumpy()  ndarray dim 1 = 640   
```

I tried 

```python
import logging
logging.basicConfig(level=logging.ERROR)
```

But this did not help ..

magnusja, 2020-03-24:
Hey there,

so there is no other way than going into the source code and deleting these lines and compiling again. #softwareengineeringatitsbest

Pendragon, 2020-12-19:
As mentioned on help screen for various tools, it's possible to control logs with a parameter.

`     param = []
        param.append("--log-level=error")
        param.append(f"--labels={config['classes']}")
        param.append(f"--colors={config['colors']}")
        param.append(f"--model={config['model']}")
        self.net = jetson.inference.segNet("fcn-resnet18-cityscapes-512x256", param)
`

magnusja, 2020-12-19:
Thanks for pointing that out! I cannot confirm that as I long moved on and not using this library anymore. But this makes sense to me so I will close the issue!

closed 2020-12-19 by magnusja

vbanchenko, 2021-04-19:
> Thanks for pointing that out! I cannot confirm that as I long moved on and not using this library anymore. But this makes sense to me so I will close the issue!

This does not work, that's why I think current issue should not have been closed.
My example:
```jetson.inference.detectNet("ssd-mobilenet-v2", ("--log-level=error", f"--threshold={INFERENCE_THRESHOLD}"))```

Output:
```detectNet -- loading detection network model from:
          -- model        networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
          -- input_blob   'Input'
          -- output_blob  'NMS'
          -- output_count 'NMS_1'
          -- class_labels networks/SSD-Mobilenet-v2/ssd_coco_labels.txt
          -- threshold    0.500000
          -- batch_size   1
```
As you can see, 'log-level' is ignored...

dusty-nv, 2021-04-19:
Can you try it as this instead?

`jetson.inference.detectNet("ssd-mobilenet-v2", arvg=["--log-level=error", f"--threshold={INFERENCE_THRESHOLD}"])`

vbanchenko, 2021-04-20:
> Can you try it as this instead?
> 
> `jetson.inference.detectNet("ssd-mobilenet-v2", arvg=["--log-level=error", f"--threshold={INFERENCE_THRESHOLD}"])`

Sure,
I've tried this, as I guess there is a typo in 'arvg':
```jetson.inference.detectNet("ssd-mobilenet-v2", argv=["--log-level=error", f"--threshold={INFERENCE_THRESHOLD}"])```
and it didn't change anything, I mean output was the same as I had mentioned earlier.

dusty-nv, 2021-04-20:
Hmm just as a test - if you just run detectnet.py with `--log-level=error`, does the setting take effect?

slyfoxneil, 2023-09-23:
try the following in python

from jetson_utils import Log

Log.SetLevel('silent')

Mitchell-Conrad-20, 2024-04-14:
> try the following in python
> 
> from jetson_utils import Log
> 
> Log.SetLevel('silent')

This is the only thing that ever worked for me. Have been looking for a solution for nearly a year. Thanks so much

nonsintetic, 2024-04-21:
Here's all the options for Log.Setlevel() in case anyone needs them:

`"silent"; "error"; "warning"; "success"; "info"; "verbose"; "debug";`

Didn't find them in the API docs so I dug through the source code.

slyfoxneil, 2024-04-21:
Hello,

It is a standard python Log library. You should find the documentation
there.

Cheers!

Neil Ablang