dusty-nv/jetson-inference #475: Demo compatibility with jetpack 4.3

rahairi, 2020-01-10:
Look like the demo cannot be run with tensorRT 6.0.1.

dusty-nv, 2020-01-10:
What error do you get?   I am able to run it with JetPack 4.3.

yahoo2016, 2020-01-12:
I was able to run detectnet-camera with Jetpack 4.2. I installed Jetpack 4.3 on TX2 Dev Kit today and verified on board camera with "camera-viewer" but could not have images displayed, , the messages are:

nvidia@l4t331:~$ detectnet-camera
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera attempting to initialize with GST_SOURCE_NVARGUS, camera 0
[gstreamer] gstCamera pipeline string:
nvarguscamerasrc sensor-id=0 ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, framerate=30/1, format=(string)NV12 ! nvvidconv flip-method=2 ! video/x-raw ! appsink name=mysink
[gstreamer] gstCamera successfully initialized with GST_SOURCE_NVARGUS, camera 0

detectnet-camera:  successfully initialized camera device
    width:  1280
   height:  720
    depth:  12 (bpp)

detectNet -- loading detection network model from:
          -- model        networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
          -- input_blob   'Input'
          -- output_blob  'NMS'
          -- output_count 'NMS_1'
          -- class_labels networks/SSD-Mobilenet-v2/ssd_coco_labels.txt
          -- threshold    0.500000
          -- batch_size   1

[TRT]   TensorRT version 6.0.1
[TRT]   loading NVIDIA plugins...
[TRT]   Plugin Creator registration succeeded - GridAnchor_TRT
[TRT]   Plugin Creator registration succeeded - GridAnchorRect_TRT
[TRT]   Plugin Creator registration succeeded - NMS_TRT
[TRT]   Plugin Creator registration succeeded - Reorg_TRT
[TRT]   Plugin Creator registration succeeded - Region_TRT
[TRT]   Plugin Creator registration succeeded - Clip_TRT
[TRT]   Plugin Creator registration succeeded - LReLU_TRT
[TRT]   Plugin Creator registration succeeded - PriorBox_TRT
[TRT]   Plugin Creator registration succeeded - Normalize_TRT
[TRT]   Plugin Creator registration succeeded - RPROI_TRT
[TRT]   Plugin Creator registration succeeded - BatchedNMS_TRT
[TRT]   Could not register plugin creator:  FlattenConcat_TRT in namespace: 
[TRT]   completed loading NVIDIA plugins.
[TRT]   detected model format - UFF  (extension '.uff')
[TRT]   desired precision specified for GPU: FASTEST
[TRT]   requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]   native precisions detected for GPU:  FP32, FP16
[TRT]   selecting fastest native precision for GPU:  FP16
[TRT]   attempting to open engine cache file /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff.1.1.GPU.FP16.engine
[TRT]   cache file not found, profiling network model on device GPU
[TRT]   device GPU, loading /usr/local/bin/ /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
[TRT]   UFFParser: Parsing Input[Op: Input].
[TRT]   UFFParser: Input -> [3,300,300]
[TRT]   UFFParser: Applying order forwarding to: Input
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/weights[Op: Const].
[TRT]   UFFParser: FeatureExtractor/MobilenetV2/Conv/weights -> [3,3,3,32]
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/weights
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/Conv2D[Op: Conv]. Inputs: Input, FeatureExtractor/MobilenetV2/Conv/weights
[TRT]   UFFParser: FeatureExtractor/MobilenetV2/Conv/Conv2D -> [32,150,150]
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/Conv2D
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/BatchNorm/moving_variance[Op: Const].
[TRT]   UFFParser: FeatureExtractor/MobilenetV2/Conv/BatchNorm/moving_variance -> [32]
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/BatchNorm/moving_variance
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/add/y[Op: Const].
[TRT]   UFFParser: FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/add/y -> []
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/add/y
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/add[Op: Binary]. Inputs: FeatureExtractor/MobilenetV2/Conv/BatchNorm/moving_variance, FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/add/y
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/add
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/Rsqrt[Op: Unary]. Inputs: FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/add
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/Rsqrt
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/BatchNorm/gamma[Op: Const].
[TRT]   UFFParser: FeatureExtractor/MobilenetV2/Conv/BatchNorm/gamma -> [32]
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/BatchNorm/gamma
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/mul[Op: Binary]. Inputs: FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/Rsqrt, FeatureExtractor/MobilenetV2/Conv/BatchNorm/gamma
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/mul
[... 10 lines cut ...]
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/mul_2
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/sub[Op: Binary]. Inputs: FeatureExtractor/MobilenetV2/Conv/BatchNorm/beta, FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/mul_2
[... cut ...]
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/expanded_conv/depthwise/BatchNorm/batchnorm/add[Op: Binary]. Inputs: FeatureExtractor/MobilenetV2/expanded_conv/depthwise/BatchNorm/moving_variance, FeatureExtractor/MobilenetV2/expanded_conv/depthwise/BatchNorm/batchnorm/add/y
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/expanded_conv/depthwise/BatchNorm/batchnorm/add
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/expanded_conv/depthwise/BatchNorm/batchnorm/Rsqrt[Op: Unary]. Inputs: FeatureExtractor/MobilenetV2/expanded_conv/depthwise/BatchNorm/batchnorm/add

rahairi, 2020-01-13:
no camera output, never ending loop on tactic

[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera attempting to initialize with GST_SOURCE_NVARGUS, camera 0
[gstreamer] gstCamera pipeline string:
nvarguscamerasrc sensor-id=0 ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, framerate=30/1, format=(string)NV12 ! nvvidconv flip-method=2 ! video/x-raw ! appsink name=mysink
[gstreamer] gstCamera successfully initialized with GST_SOURCE_NVARGUS, camera 0

detectnet-camera:  successfully initialized camera device
    width:  1280
   height:  720
    depth:  12 (bpp)

detectNet -- loading detection network model from:
          -- model        networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
          -- input_blob   'Input'
          -- output_blob  'NMS'
          -- output_count 'NMS_1'
          -- class_labels networks/SSD-Mobilenet-v2/ssd_coco_labels.txt
          -- threshold    0.500000
          -- batch_size   1

[TRT]   TensorRT version 6.0.1
[TRT]   loading NVIDIA plugins...
[TRT]   Plugin Creator registration succeeded - GridAnchor_TRT
[TRT]   Plugin Creator registration succeeded - GridAnchorRect_TRT
[TRT]   Plugin Creator registration succeeded - NMS_TRT
[TRT]   Plugin Creator registration succeeded - Reorg_TRT
[TRT]   Plugin Creator registration succeeded - Region_TRT
[TRT]   Plugin Creator registration succeeded - Clip_TRT
[TRT]   Plugin Creator registration succeeded - LReLU_TRT
[TRT]   Plugin Creator registration succeeded - PriorBox_TRT
[TRT]   Plugin Creator registration succeeded - Normalize_TRT
[TRT]   Plugin Creator registration succeeded - RPROI_TRT
[TRT]   Plugin Creator registration succeeded - BatchedNMS_TRT
[TRT]   Could not register plugin creator:  FlattenConcat_TRT in namespace: 
[TRT]   completed loading NVIDIA plugins.
[TRT]   detected model format - UFF  (extension '.uff')
[TRT]   desired precision specified for GPU: FASTEST
[TRT]   requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]   native precisions detected for GPU:  FP32, FP16
[TRT]   selecting fastest native precision for GPU:  FP16
[TRT]   attempting to open engine cache file networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff.1.1.GPU.FP16.engine
[TRT]   cache file not found, profiling network model on device GPU
[TRT]   device GPU, loading /usr/local/bin/ networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
[TRT]   UFFParser: Parsing Input[Op: Input].
[TRT]   UFFParser: Input -> [3,300,300]
[TRT]   UFFParser: Applying order forwarding to: Input
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/weights[Op: Const].
[TRT]   UFFParser: FeatureExtractor/MobilenetV2/Conv/weights -> [3,3,3,32]
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/weights
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/Conv2D[Op: Conv]. Inputs: Input, FeatureExtractor/MobilenetV2/Conv/weights
[TRT]   UFFParser: FeatureExtractor/MobilenetV2/Conv/Conv2D -> [32,150,150]
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/Conv2D
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/BatchNorm/moving_variance[Op: Const].
[TRT]   UFFParser: FeatureExtractor/MobilenetV2/Conv/BatchNorm/moving_variance -> [32]
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/BatchNorm/moving_variance
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/add/y[Op: Const].
[TRT]   UFFParser: FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/add/y -> []
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/add/y
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/add[Op: Binary]. Inputs: FeatureExtractor/MobilenetV2/Conv/BatchNorm/moving_variance, FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/add/y
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/add
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/Rsqrt[Op: Unary]. Inputs: FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/add
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/Rsqrt
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/BatchNorm/gamma[Op: Const].
[TRT]   UFFParser: FeatureExtractor/MobilenetV2/Conv/BatchNorm/gamma -> [32]
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/BatchNorm/gamma
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/mul[Op: Binary]. Inputs: FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/Rsqrt, FeatureExtractor/MobilenetV2/Conv/BatchNorm/gamma
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/mul
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/mul_1[Op: Binary]. Inputs: FeatureExtractor/MobilenetV2/Conv/Conv2D, FeatureExtractor/MobilenetV2/Conv/BatchNorm/batchnorm/mul
[... 6 lines cut ...]
[TRT]   UFFParser: FeatureExtractor/MobilenetV2/Conv/BatchNorm/moving_mean -> [32]
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/BatchNorm/moving_mean
[... cut ...]
[TRT]   UFFParser: FeatureExtractor/MobilenetV2/expanded_conv/depthwise/BatchNorm/moving_variance -> [32]
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/expanded_conv/depthwise/BatchNorm/moving_variance
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/expanded_conv/depthwise/BatchNorm/batchnorm/add/y[Op: Const].
[TRT]   UFFParser: FeatureExtractor/MobilenetV2/expanded_conv/depthwise/BatchNorm/batchnorm/add/y -> []
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/expanded_conv/depthwise/BatchNorm/batchnorm/add/y

rahairi, 2020-01-13:
+ FeatureExtractor/MobilenetV2/Conv/Relu6 (CudaDepthwiseConvolution)
[TRT]   CudaDepthwiseConvolution has no valid tactics for this config, skipping
[TRT]   >>>>>>>>>>>>>>> Chose Runner Type: CaskConvolution Tactic: 1062367460111450758
[TRT]   FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (scudnn) Set Tactic Name: maxwell_scudnn_128x32_relu_medium_nn_v1
[TRT]   
[TRT]   *************** Autotuning format combination: Half(1,300,90000,270000) -> Half(1,150,22500,720000) ***************
[TRT]   --------------- Timing Runner: FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (FusedConvActConvolution)
[TRT]   FusedConvActConvolution has no valid tactics for this config, skipping
[TRT]   --------------- Timing Runner: FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (CaskConvolution)
[TRT]   CaskConvolution has no valid tactics for this config, skipping
[TRT]   --------------- Timing Runner: FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (CudaConvolution)
[TRT]   Tactic: 0 time 2.02477
[TRT]   Tactic: 1 time 1.59526
[TRT]   Tactic: 2 time 2.3631
[TRT]   Fastest Tactic: 1 Time: 1.59526
[TRT]   --------------- Timing Runner: FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (CudaDepthwiseConvolution)
[TRT]   CudaDepthwiseConvolution has no valid tactics for this config, skipping
[TRT]   >>>>>>>>>>>>>>> Chose Runner Type: CudaConvolution Tactic: 1
[TRT]   
[TRT]   *************** Autotuning format combination: Half(1,300,90000:2,180000) -> Half(1,150,22500:2,360000) ***************
[TRT]   --------------- Timing Runner: FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (LegacySASSConvolution)
[TRT]   LegacySASSConvolution has no valid tactics for this config, skipping
[TRT]   --------------- Timing Runner: FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (FusedConvActConvolution)
[TRT]   FusedConvActConvolution has no valid tactics for this config, skipping
[TRT]   --------------- Timing Runner: FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (CaskConvolution)
[TRT]   FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (hcudnn) Set Tactic Name: maxwell_fp16x2_hcudnn_fp16x2_128x32_relu_medium_nn_v1
[TRT]   Tactic: 3564772625446233998 time 0.55362
[TRT]   FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (hcudnn) Set Tactic Name: maxwell_fp16x2_hcudnn_fp16x2_128x32_relu_large_nn_v1
[TRT]   Tactic: 3650389455493082349 time 0.581198
[TRT]   FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (hcudnn) Set Tactic Name: maxwell_fp16x2_hcudnn_fp16x2_128x32_relu_small_nn_v1
[TRT]   Tactic: 5319956359050645452 time 0.52362
[TRT]   FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (hcudnn) Set Tactic Name: maxwell_fp16x2_hcudnn_fp16x2_128x64_relu_medium_nn_v1
[TRT]   Tactic: 7205456024582378848 time 0.810052
[TRT]   FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (hcudnn) Set Tactic Name: maxwell_fp16x2_hcudnn_fp16x2_128x64_relu_large_nn_v1
[TRT]   Tactic: -6490690591794140522 time 0.85289
[TRT]   FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (hcudnn) Set Tactic Name: maxwell_fp16x2_hcudnn_fp16x2_128x128_relu_large_nn_v1
[TRT]   Tactic: -4686027666808657977 time 1.62779
[TRT]   FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (hcudnn) Set Tactic Name: maxwell_fp16x2_hcudnn_fp16x2_128x128_relu_small_nn_v1
[TRT]   Tactic: -4212163711445252890 time 1.59414
[TRT]   FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (hcudnn) Set Tactic Name: maxwell_fp16x2_hcudnn_fp16x2_128x128_relu_medium_nn_v1
[TRT]   Tactic: -3898373634979201110 time 1.59753
[TRT]   FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (hcudnn) Set Tactic Name: maxwell_fp16x2_hcudnn_fp16x2_128x64_relu_small_nn_v1
[TRT]   Tactic: -2409163523992614473 time 0.808854
[TRT]   Fastest Tactic: 5319956359050645452 Time: 0.52362
[TRT]   --------------- Timing Runner: FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (CudaConvolution)
[TRT]   CudaConvolution has no valid tactics for this config, skipping
[TRT]   --------------- Timing Runner: FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (CudaDepthwiseConvolution)
[TRT]   CudaDepthwiseConvolution has no valid tactics for this config, skipping
[TRT]   >>>>>>>>>>>>>>> Chose Runner Type: CaskConvolution Tactic: 5319956359050645452
[TRT]   FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6 (hcudnn) Set Tactic Name: maxwell_fp16x2_hcudnn_fp16x2_128x32_relu_small_nn_v1
[TRT]   
[TRT]   --------------- Timing Runner: <reformat> (Reformat)
[TRT]   Tactic: 1002 time 0.548594
[TRT]   Tactic: 0 time 0.799531
[TRT]   Fastest Tactic: 1002 Time: 0.548594
[TRT]   --------------- Timing Runner: <reformat> (Reformat)
[TRT]   Tactic: 1002 time 0.535573
[TRT]   Tactic: 0 time 0.678542
[TRT]   Fastest Tactic: 1002 Time: 0.535573
[TRT]   --------------- Timing Runner: <reformat> (Reformat)
[TRT]   Tactic: 1002 time 2.16971
[TRT]   Tactic: 0 time 0.58461
[... cut ...]
[TRT]   Tactic: 53 time 1.46833
[TRT]   Tactic: 55 time 2.09742
[TRT]   Tactic: 59 time 2.24919
[TRT]   Tactic: 60 time 1.54326
[TRT]   Tactic: 61 time 4.2582
[TRT]   Tactic: 63 time 1.46641
[TRT]   Tactic: 64 time 1.9362
[TRT]   Tactic: 65 time 1.34849
[TRT]   Tactic: 68 time 1.15237
[TRT]   Tactic: 69 time 1.54841
[TRT]   Tactic: 70 time 2.54453
[TRT]   Tactic: 74 time 1.56318
[TRT]   Tactic: 75 time 1.29628
[TRT]   Tactic: 78 time 2.24625
[TRT]   Tactic: 79 time 1.63375
[TRT]   Tactic: 81 time 2.37049

dusty-nv, 2020-01-13:
The SSD models take a few minutes to load the first time (around 5 minutes).  I just tried it again here, and it loads and runs as expected on JetPack 4.3.

SaddamBInSyed, 2020-01-16:
@dusty-nv ,

I bought my jetson nano board.

Could you please explain what is the difference between Jetson inference SDK object detection code vs Deepstream SDk4.0.2 object detection code?

when should we go for what?

highly appreciated your explanation.

Philliec459, 2020-05-27:
Can I ask what actually is going on the first time you try to run a model vs. every time after that? The first time I tried a few models with my web cam it seemed to go on and on so I killed the program thinking that I had the wrong webcam. Now it works perfect. I would just like to know what the program does the fist time that is do different. Thanks

WyattAutomation, 2020-05-30:
Confirmed that this is the case with running the python detectnet-camera.py example on the Nano; both the coco and the SSD models had to run for 8 and 5 minutes respectively on the loop of 'tactic' information that whizzes by on their first load.  Subsequent uses start up right away.

Might be worth a quick mention in the documentation for the examples.  I did the same thing and fuddled around with swapping/testing cams before I found this thread; wasn't expecting that to be the issue but I'm glad it works and appreciate the help here!

closed 2023-04-27 by dusty-nv