dusty-nv/jetson-inference #478: Jetson inference vs Deepstream 4.0.2 performance

SaddamBInSyed, 2020-01-16:
@dusty-nv ,

I bought my jetson nano board.

Could you please explain what is the difference between Jetson inference SDK object detection code vs Deepstream SDk4.0.2 object detection code?

when should we go for what?

highly appreciated your explanation.

dusty-nv, 2020-01-16:
Hi @SaddamBInSyed , DeepStream framework is optimized for multi-stream operations.  `jetson-inference` has other types of DNNs available like image recognition, semantic segmentation, and others.  You can start with `jetson-inference` and move to DeepStream if you require deploying better performance across many streams.

SaddamBInSyed, 2020-01-20:
Thank you, @dusty-nv.
since I am handling 2 camera I will go with deepstream 4.0.2 SDK.
I installed the Deep stream sdk on my jetson nano.
but when I run the object detection code (SSD) sample with a 2IP camera as an argument, 
its very slow like 4 FPS.

How do I get the good FPS?

please advise.

dusty-nv, 2020-01-23:
Hi @SaddamBInSyed, please post technical questions about DeepStream SDK to here:  https://devtalk.nvidia.com/default/board/209/deepstream-sdk/

closed 2020-01-28 by SaddamBInSyed