dusty-nv/jetson-inference #479: Custom models (PyTorch, ONNX, Caffe)

theDweeb, 2020-01-17:
Hello, I am currently trying to run custom models on the Nvidia Xavier using this jetson-inference repo. I have gone through the tutorials using the prebuilt networks just fine but when I try to use custom models I run into problems.

### Pretrained MobileNetV2 model from Torchvision

I converted it to ONNX format using [this](https://pytorch.org/tutorials/advanced/super_resolution_with_onnxruntime.html)

`Exported model has been tested with ONNXRuntime, and the result looks good!
`

but fails the TensorRT onnx test:
```
/usr/src/tensorrt/bin/trtexec --onnx=onnx_models/MobileNetV2.onnx 
&&&& RUNNING TensorRT.trtexec # /usr/src/tensorrt/bin/trtexec --onnx=onnx_models/MobileNetV2.onnx
[01/17/2020-11:34:07] [I] === Model Options ===
[01/17/2020-11:34:07] [I] Format: ONNX
[01/17/2020-11:34:07] [I] Model: onnx_models/MobileNetV2.onnx
[01/17/2020-11:34:07] [I] Output:
[01/17/2020-11:34:07] [I] === Build Options ===
[01/17/2020-11:34:07] [I] Max batch: 1
[01/17/2020-11:34:07] [I] Workspace: 16 MB
[01/17/2020-11:34:07] [I] minTiming: 1
[01/17/2020-11:34:07] [I] avgTiming: 8
[01/17/2020-11:34:07] [I] Precision: FP32
[01/17/2020-11:34:07] [I] Calibration: 
[01/17/2020-11:34:07] [I] Safe mode: Disabled
[01/17/2020-11:34:07] [I] Save engine: 
[01/17/2020-11:34:07] [I] Load engine: 
[01/17/2020-11:34:07] [I] Inputs format: fp32:CHW
[01/17/2020-11:34:07] [I] Outputs format: fp32:CHW
[01/17/2020-11:34:07] [I] Input build shapes: model
[01/17/2020-11:34:07] [I] === System Options ===
[01/17/2020-11:34:07] [I] Device: 0
[01/17/2020-11:34:07] [I] DLACore: 
[01/17/2020-11:34:07] [I] Plugins:
[01/17/2020-11:34:07] [I] === Inference Options ===
[01/17/2020-11:34:07] [I] Batch: 1
[01/17/2020-11:34:07] [I] Iterations: 10
[01/17/2020-11:34:07] [I] Duration: 3s (+ 200ms warm up)
[01/17/2020-11:34:07] [I] Sleep time: 0ms
[01/17/2020-11:34:07] [I] Streams: 1
[01/17/2020-11:34:07] [I] ExposeDMA: Disabled
[01/17/2020-11:34:07] [I] Spin-wait: Disabled
[01/17/2020-11:34:07] [I] Multithreading: Disabled
[01/17/2020-11:34:07] [I] CUDA Graph: Disabled
[01/17/2020-11:34:07] [I] Skip inference: Disabled
[01/17/2020-11:34:07] [I] Input inference shapes: model
[01/17/2020-11:34:07] [I] Inputs:
[01/17/2020-11:34:07] [I] === Reporting Options ===
[01/17/2020-11:34:07] [I] Verbose: Disabled
[01/17/2020-11:34:07] [I] Averages: 10 inferences
[01/17/2020-11:34:07] [I] Percentile: 99
[01/17/2020-11:34:07] [I] Dump output: Disabled
[01/17/2020-11:34:07] [I] Profile: Disabled
[01/17/2020-11:34:07] [I] Export timing to JSON file: 
[01/17/2020-11:34:07] [I] Export output to JSON file: 
[01/17/2020-11:34:07] [I] Export profile to JSON file: 
[01/17/2020-11:34:07] [I] 
----------------------------------------------------------------
Input filename:   onnx_models/MobileNetV2.onnx
ONNX IR version:  0.0.4
Opset version:    10
Producer name:    pytorch
Producer version: 1.2
Domain:           
Model version:    0
Doc string:       
----------------------------------------------------------------
[01/17/2020-11:34:08] [W] [TRT] Calling isShapeTensor before the entire network is constructed may result in an inaccurate result.
[01/17/2020-11:34:08] [E] [TRT] Network has dynamic or shape inputs, but no optimization profile has been defined.
[01/17/2020-11:34:08] [E] [TRT] Network validation failed.
[... 193 lines cut ...]
Error: Weight 99977.8 is outside of [65504, -65504].
Error: Weight 99977.8 is outside of [65504, -65504].
Error: Weight 99977.8 is outside of [65504, -65504].
Error: Weight 99977.8 is outside of [65504, -65504].
[TRT]   device GPU, failed to parse caffe network
[TRT]   device GPU, failed to load MobileNet-Caffe/mobilenet_v2.caffemodel
[TRT]   failed to load MobileNet-Caffe/mobilenet_v2.caffemodel
[TRT]   imageNet -- failed to initialize.
jetson.inference -- imageNet failed to load built-in network 'googlenet'
PyTensorNet_Dealloc()
Traceback (most recent call last):
  File "./imagenet-console.py", line 49, in <module>
    net = jetson.inference.imageNet(opt.network, sys.argv)
Exception: jetson.inference -- imageNet failed to load network
jetson.utils -- freeing CUDA mapped memory

```

<hr>

I know jetson-inference supplies a prebuilt MobileNetV2 + SSD but I am trying to get a custom mobilenetV2 model (personally trained) to work but decided to see if I could get the original to work first (above models).

Ideally I would like to be able to get my custom PyTorch models converted to ONNX (or native PyTorch support?) and run them on the Xavier.

Thank you for any and all help.

theDweeb, 2020-01-23:
Update: I got my custom model working. The issue with converting PyTorch -> ONNX was some distiller packages for quantization that ONNX did not like. Also, I was setting the --input_blob and output_blob flags to the same name as the ones in the tutorial instead of my ONNX model's input/output name.

closed 2020-01-23 by theDweeb

zoeyhl, 2021-06-23:
Hi, could you pls explain what you mean by "ONNX model's input/output name"?