dusty-nv/jetson-inference #480: Flipped output frames.

DoriHp, 2020-01-20:
Thanks for great work. I tried your repo, everything work fine, except one. When program show processed frame, it was flipped like this: 
[image]

But when I try this repo: https://github.com/NVIDIA-AI-IOT/jetcam, which also read frame using gstreamer, that error didn't happend. How can I resolve it?

DoriHp, 2020-01-20:
@dusty-nv

dusty-nv, 2020-01-23:
Hi @DoriHp , try changing the `flip-method` parameter to 0 here:  https://github.com/dusty-nv/jetson-utils/blob/798c416c175d509571859c9290257bd5cce1fd63/camera/gstCamera.cpp#L416

Then build/install jetson-inference again:

```bash
cd /path/to/your/jetson-inference/build
make
sudo make install
```

DoriHp, 2020-01-30:
Thank for your reply

closed 2020-01-30 by DoriHp

AI-Guru, 2020-02-02:
Would it be possible to have flip-method as a parameter for instantiating the camera? This way no one would have to recompile :D