dusty-nv/jetson-inference #482: Cmake not Detecting Virtual Environment (venv)

navyjosh, 2020-01-21:
I'm using Jetson Nano, following the build from source instructions. However, I decided to use a venv.

I'm using virtualenv with virtualenvwrapper and when I run cmake, it is only installing python modules to /usr/lib/python* whereas my venv is not being detected and no python modules are being installed to the venv dir.

I made sure to activate the venv, and I check that the venv is in $PATH.

dusty-nv, 2020-02-18:
Hi @navyjosh , you could try changing the `PYTHON_BINDING_INSTALL_DIR` here:   https://github.com/dusty-nv/jetson-inference/blob/8846dcb274ff812880ef7bf9d42eb1dd75c9e7b7/python/bindings/CMakeLists.txt#L52

Kmarconi, 2020-04-16:
Hi,

Sorry for re-opening this issue but even after making the change in the python binding CmakeList file (and compiling from source successfully) , I'm still not able to import the jetson module in my virtualenv's python. Any other file which should change ?

Regards

shubhamsidhwa, 2020-06-12:
Was this issue resolved  ? I am facing the same issue

Kmarconi, 2020-06-12:
If I remember well, I had to add the lib path my bashrc file :)

shubhamsidhwa, 2020-06-12:
Which bashrc file ?

shubhamsidhwa, 2020-06-12:
set(PYTHON_BINDING_INSTALL_DIR /usr/lib/python${PYTHON_VERSION_MAJOR}.${PYTHON_VERSION_MINOR}/dist-packages)  : I did this change as suggested by dusty-nv above

iliesaya, 2021-02-23:
don't forget to also update the file : utils/python/bindings/CMakeLists.txt
in my case I replaced with : set(PYTHON_BINDING_INSTALL_DIR /home/jetson/.virtualenvs/py3flask/lib/python3.6/site-packages)

abbasitaher, 2021-04-01:
I changed the python binding CmakeList file and everything went okay but when I imported jetson.utils this error occurs:

import jetson.utils
  File "path/to/venv/lib/python3.6/site-packages/jetson/utils/__init__.py", line 4, in <module>
    from jetson_utils_python import *
ImportError: dynamic module does not define module export function (PyInit_jetson_utils_python)

abbasitaher, 2021-04-01:
I managed to solve the problem. Remove or rename jetson_utils_python and jetson_inference_python.so under path/to/venv/lib/python3.6/site-packages. Then:
$ cd path/to/venv/lib/python3.6/site-packages
$ sudo ln -s /usr/lib/python3.6/dist-packages/jetson_inference_python.so
$ sudo ln -s /usr/lib/python3.6/dist-packages/jetson_utils_python.so

ps: if you couldn't find the files under  /usr/lib/ try  /usr/local/lib/

closed 2021-08-18 by navyjosh