dusty-nv/jetson-inference #483: Collecting your own dataset tutorial won't run

socd06, 2020-01-22:
Link to tutorial:
[Collecting your own dataset](https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect.md)

I've been working on this for three days and nothing seems to work.

I collected my own data, trained for four hours and whenever I try using the model (wether on images or on the camera) I get this message:

> [TRT]   TensorRT version 5.1.6
[TRT]   loading NVIDIA plugins...
[TRT]   Plugin Creator registration succeeded - GridAnchor_TRT
[TRT]   Plugin Creator registration succeeded - NMS_TRT
[TRT]   Plugin Creator registration succeeded - Reorg_TRT
[TRT]   Plugin Creator registration succeeded - Region_TRT
[TRT]   Plugin Creator registration succeeded - Clip_TRT
[TRT]   Plugin Creator registration succeeded - LReLU_TRT
[TRT]   Plugin Creator registration succeeded - PriorBox_TRT
[TRT]   Plugin Creator registration succeeded - Normalize_TRT
[TRT]   Plugin Creator registration succeeded - RPROI_TRT
[TRT]   Plugin Creator registration succeeded - BatchedNMS_TRT
[TRT]   completed loading NVIDIA plugins.
[TRT]   detected model format - ONNX  (extension '.onnx')
[TRT]   desired precision specified for GPU: FASTEST
[TRT]   requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]   native precisions detected for GPU:  FP32, FP16
[TRT]   selecting fastest native precision for GPU:  FP16
[TRT]   attempting to open engine cache file .1.1.GPU.FP16.engine
[TRT]   cache file not found, profiling network model on device GPU

> error:  model file '~/datasets/small-fruit5/resnet18.onnx' was not found. 
>  if loading a built-in model, maybe it wasn't downloaded before.
> 
> Run the Model Downloader tool again and select it for download:
>            $ cd <jetson-inference>/tools
>            $ ./download-models.sh
> [TRT]   failed to load ~/datasets/small-fruit5/resnet18.onnx
> [TRT]   imageNet -- failed to initialize.
jetson.inference -- imageNet failed to load built-in network 'googlenet'
PyTensorNet_Dealloc()
Traceback (most recent call last):
  File "/usr/local/bin/imagenet-console.py", line 49, in <module>
    net = jetson.inference.imageNet(opt.network, sys.argv)
Exception: jetson.inference -- imageNet failed to load network
jetson.utils -- freeing CUDA mapped memory

I also tried running the onnx_validate.py script but it gives me an "insufficient arguments" error.

dusty-nv, 2020-01-22:
Hi @socd06 , what is the command line you are running the imagenet- program with?

You might want to try specifying the full path instead of `~` (ie `/home/myuser/datasets/small-fruit5/resnet18.onnx`)

socd06, 2020-01-22:
I use the full path every time. I go to jetson-inference/python/examples
and then run imagenet-camera

The path I use looks something like

~/jetson-inference/python/training/classification/small-fruit5/

That runtime I was trying putting the onnx file somewhere with a shorter
path.

You can see a little bit above that it says it recognizes the onnx format,
then looks for the fp16 cache file and then "network not found".

I've been fiddling with it for a few days now and it just does this with
custom models.

I also can't even run onnx_validate.py, there are no examples and every
model path I put I get "insufficient arguments" or a runtime error.

dusty-nv, 2020-01-22:
By full path, I mean without the '~' (home directory shortcut).  Instead use the absolute path to your home directory instead (e.g. `/home/myuser/datasets/small-fruit5/resnet18.onnx`)   My thought is that Python may require explicit expansion of `~` in the code, which may not be done.

Can you also please provide the command line you are running?

socd06, 2020-01-23:
Hi @dusty-nv 

My command line was:

> imagenet-console.py --model=~/datasets/small-fruit5/resnet18.onnx --input_blob=input_0 --output_blob=output_0 --labels=~/datasets/small-fruit5/labels.txt ~/Downloads/stb.jpg

I tried what you said and it's now working
the command line now looks like this: 

> imagenet-console.py --model=/home/carlos/datasets/small-fruit5/resnet18.onnx --input_blob=input_0 --output_blob=output_0 --labels=/home/carlos/datasets/small-fruit5/labels.txt /home/carlos/Downloads/stb.jpg

This was super frustrating for me being rusty with my Ubuntu, hope you guys can add a note or an example to the tutorials.

Thanks a lot

closed 2020-01-23 by socd06