dusty-nv/jetson-inference #484: imagenet-camera does not classify using the cat_dog_epoch_100 model

hyl-g, 2020-01-23:
I followed the [instruction](https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-cat-dog.md) to download the cat_dog_epoch_100 model and convert it to onnx.  I ran the imagenet-camera using this model but it did not classify anything.  I ran the imagnet-camera using the default GoogleNet and it worked fine.

I'm wondering if there anybody have solution to make imagenet-camera works for custom onnx models.

Thanks,

dusty-nv, 2020-01-23:
Hi @hyl-g, what is the command line that you ran the cat_dog_epoch_100 onnx model with?

On your system the cat_dog_epoch_100 model able to classify any static images using `imagenet-console` program?

Can you provide the output of a run of `imagenet-console` with it please?  Thanks.

hyl-g, 2020-01-23:
Hi Dusty,

Thanks for you reply and thanks for your great work to put together this package for the other developers.

Yes, the image-console works but for some reason, the image-camera doesn't.  Below are my command lines.  The model is in folder cat_dog_epoch:
_DATASET=~/datasets/cat_dog
imagenet-camera --model=cat_dog_epoch_100/resnet18.onnx --input_blob=input_0 --output_blob=output_0 --labels=$DATASET/labels.txt --camera=/dev/video0 --width=352 --height=288_

What I don't understand is only the combination of imagenet-camera and custom onnx model has problem.  Other combinations such as imagenet-console + custom onnx model and imagenet-camera + GoogleNet work fine.

Regards
Henry

dusty-nv, 2020-01-23:
Hi Henry, what is it viewing on the camera when you run the cat_dog_epoch_100 on imagenet-camera?  

The cat/dog models support 2 classes (cat and dog).  So camera would probably need to be looking at cat or dog to classify correctly.  What does the camera program classify as when the camera is pointed at nothing?   Or does the app not even run?

hyl-g, 2020-01-23:
I hold up a cat picture (displayed in my cell phone) in front of the camera.  The jetson display showed the picture but the code says dog.  I tried many another cat pictures.  The answers were dog.  The system is not stuck.  I see the confidence/FPS number keep updating but It seems that the output always infer a dog.  This is the way I test the GoogleNet.  GoogleNet recognizes the cat correctly.

dusty-nv, 2020-01-23:
Hmm - do the confidence values for dog change as you try different images on your phone?

Since the cat_dog model is working with imagenet-console, I think it is working, that model is just not doing very well on your phone images it seems.

hyl-g, 2020-01-23:
When tested the attached images with the imagenet-console, I got:
50% dog for cat1
66% cat for cat2
97% cat for cat3
I displayed the images on my cell phone and recognized it by imagenet-camera.  I got:
90% dog for cat1
66% dog for cat2
60% dog for cat3

I'm wondering if you can repeat the test on your end to see if you get the same result.

Thanks,
Henry
[image]
[image]
[image]

dipampatel18, 2020-02-04:
@hyl-g As @dusty-nv mentioned, the issue is because of the images present on your phone and maybe the model doesn't perform that well when getting data from a phone instead of looking at an object directly.

I think the resolution is dropping on getting the images from the phone. Try displaying the same images on your PC instead and I believe it should give satisfactory results.

hyl-g, 2020-02-07:
Thanks for your tips.  Try that but the result is not better (see below):
cat 1 image: 80% dog
cat 2 image: 70% dog
cat3 image: 80% dog