dusty-nv/jetson-inference #486: Couldn't install jetson inference on my jetson nano+ jetpack4.3

SaddamBInSyed, 2020-01-27:
Hi @dusty-nv ,

I bought a new jetson nano and carefully configured with jetpack4.3 and when  I install the jetson-inference sdk then I am getting the below error.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/deeplearning/jetson-inference/build
deeplearning@deeplearning-desktop:~/jetson-inference/build$ make
[  1%] Linking CXX shared library ../aarch64/lib/libjetson-utils.so
/usr/bin/ld: cannot find -lGL
collect2: error: ld returned 1 exit status
utils/CMakeFiles/jetson-utils.dir/build.make:2627: recipe for target 'aarch64/lib/libjetson-utils.so' failed
make[2]: *** [aarch64/lib/libjetson-utils.so] Error 1
CMakeFiles/Makefile2:927: recipe for target 'utils/CMakeFiles/jetson-utils.dir/all' failed
make[1]: *** [utils/CMakeFiles/jetson-utils.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

Highly appreciated your prompt response.

Thanks

dusty-nv, 2020-01-27:
Had you installed or upgraded other packages before installing `jetson-inference`?  It appears that something overwrote your libGL.so

I've heard that installing this repo can fix it:  https://github.com/NVIDIA/libglvnd

Here is what my libGL binaries look like on JetPack 4.3 / Nano:

```bash
$ ls -ll /usr/lib/aarch64-linux-gnu/libGL*
lrwxrwxrwx 1 root root      22 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLdispatch.so -> libGLdispatch.so.0.0.0
lrwxrwxrwx 1 root root      22 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLdispatch.so.0 -> libGLdispatch.so.0.0.0
-rw-r--r-- 1 root root 1030368 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLdispatch.so.0.0.0
lrwxrwxrwx 1 root root      21 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLESv1_CM.so -> libGLESv1_CM.so.1.0.0
lrwxrwxrwx 1 root root      21 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLESv1_CM.so.1 -> libGLESv1_CM.so.1.0.0
-rw-r--r-- 1 root root  141472 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLESv1_CM.so.1.0.0
lrwxrwxrwx 1 root root      18 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLESv2.so -> libGLESv2.so.2.0.0
lrwxrwxrwx 1 root root      18 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLESv2.so.2 -> libGLESv2.so.2.0.0
-rw-r--r-- 1 root root  153760 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLESv2.so.2.0.0
lrwxrwxrwx 1 root root      19 Feb 22  2019 /usr/lib/aarch64-linux-gnu/libGLEWmx.so.1.13 -> libGLEWmx.so.1.13.0
-rw-r--r-- 1 root root  505624 Feb 22  2019 /usr/lib/aarch64-linux-gnu/libGLEWmx.so.1.13.0
lrwxrwxrwx 1 root root      16 Aug 24  2017 /usr/lib/aarch64-linux-gnu/libGLEW.so -> libGLEW.so.2.0.0
lrwxrwxrwx 1 root root      16 Aug 24  2017 /usr/lib/aarch64-linux-gnu/libGLEW.so.2.0 -> libGLEW.so.2.0.0
-rw-r--r-- 1 root root  616336 Aug 24  2017 /usr/lib/aarch64-linux-gnu/libGLEW.so.2.0.0
lrwxrwxrwx 1 root root      14 May 10  2019 /usr/lib/aarch64-linux-gnu/libGL.so -> libGL.so.1.0.0
lrwxrwxrwx 1 root root      14 May 10  2019 /usr/lib/aarch64-linux-gnu/libGL.so.1 -> libGL.so.1.0.0
-rw-r--r-- 1 root root  972968 May 10  2019 /usr/lib/aarch64-linux-gnu/libGL.so.1.0.0
-rw-r--r-- 1 root root  850714 May 21  2016 /usr/lib/aarch64-linux-gnu/libGLU.a
lrwxrwxrwx 1 root root      15 May 21  2016 /usr/lib/aarch64-linux-gnu/libGLU.so -> libGLU.so.1.3.1
lrwxrwxrwx 1 root root      15 May 21  2016 /usr/lib/aarch64-linux-gnu/libGLU.so.1 -> libGLU.so.1.3.1
-rw-r--r-- 1 root root  400040 May 21  2016 /usr/lib/aarch64-linux-gnu/libGLU.so.1.3.1
lrwxrwxrwx 1 root root      16 Jul 19  2019 /usr/lib/aarch64-linux-gnu/libGLX_indirect.so.0 -> libGLX_mesa.so.0
lrwxrwxrwx 1 root root      20 Jul 19  2019 /usr/lib/aarch64-linux-gnu/libGLX_mesa.so.0 -> libGLX_mesa.so.0.0.0
-rw-r--r-- 1 root root  442984 Jul 19  2019 /usr/lib/aarch64-linux-gnu/libGLX_mesa.so.0.0.0
lrwxrwxrwx 1 root root      15 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLX.so -> libGLX.so.0.0.0
lrwxrwxrwx 1 root root      15 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLX.so.0 -> libGLX.so.0.0.0
-rw-r--r-- 1 root root   63880 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLX.so.0.0.0
```

You may want to check the file sizes and symlinks against yours.

SaddamBInSyed, 2020-01-27:
yes. I have purge my opencv4 library after that I tried to install jetson inference.

below are the result.

**ls -ll /usr/lib/aarch64-linux-gnu/libGL***
lrwxrwxrwx 1 root root      22 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLdispatch.so -> libGLdispatch.so.0.0.0
lrwxrwxrwx 1 root root      22 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLdispatch.so.0 -> libGLdispatch.so.0.0.0
-rw-r--r-- 1 root root 1030368 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLdispatch.so.0.0.0
lrwxrwxrwx 1 root root      21 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLESv1_CM.so -> libGLESv1_CM.so.1.0.0
lrwxrwxrwx 1 root root      21 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLESv1_CM.so.1 -> libGLESv1_CM.so.1.0.0
-rw-r--r-- 1 root root  141472 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLESv1_CM.so.1.0.0
lrwxrwxrwx 1 root root      18 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLESv2.so -> libGLESv2.so.2.0.0
lrwxrwxrwx 1 root root      18 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLESv2.so.2 -> libGLESv2.so.2.0.0
-rw-r--r-- 1 root root  153760 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLESv2.so.2.0.0
lrwxrwxrwx 1 root root      19 Feb 22  2019 /usr/lib/aarch64-linux-gnu/libGLEWmx.so.1.13 -> libGLEWmx.so.1.13.0
-rw-r--r-- 1 root root  505624 Feb 22  2019 /usr/lib/aarch64-linux-gnu/libGLEWmx.so.1.13.0
lrwxrwxrwx 1 root root      16 Aug 24  2017 /usr/lib/aarch64-linux-gnu/libGLEW.so -> libGLEW.so.2.0.0
lrwxrwxrwx 1 root root      16 Aug 24  2017 /usr/lib/aarch64-linux-gnu/libGLEW.so.2.0 -> libGLEW.so.2.0.0
-rw-r--r-- 1 root root  616336 Aug 24  2017 /usr/lib/aarch64-linux-gnu/libGLEW.so.2.0.0
lrwxrwxrwx 1 root root      41 Jan 24 07:54 /usr/lib/aarch64-linux-gnu/libGL.so -> /usr/lib/aarch64-linux-gnu/tegra/libGL.so
lrwxrwxrwx 1 root root      14 May 10  2019 /usr/lib/aarch64-linux-gnu/libGL.so.1 -> libGL.so.1.0.0
-rw-r--r-- 1 root root  972968 May 10  2019 /usr/lib/aarch64-linux-gnu/libGL.so.1.0.0
-rw-r--r-- 1 root root  850714 May 22  2016 /usr/lib/aarch64-linux-gnu/libGLU.a
lrwxrwxrwx 1 root root      15 May 22  2016 /usr/lib/aarch64-linux-gnu/libGLU.so -> libGLU.so.1.3.1
lrwxrwxrwx 1 root root      15 May 22  2016 /usr/lib/aarch64-linux-gnu/libGLU.so.1 -> libGLU.so.1.3.1
-rw-r--r-- 1 root root  400040 May 22  2016 /usr/lib/aarch64-linux-gnu/libGLU.so.1.3.1
lrwxrwxrwx 1 root root      16 Nov 18 22:28 /usr/lib/aarch64-linux-gnu/libGLX_indirect.so.0 -> libGLX_mesa.so.0
lrwxrwxrwx 1 root root      20 Nov 18 22:28 /usr/lib/aarch64-linux-gnu/libGLX_mesa.so.0 -> libGLX_mesa.so.0.0.0
-rw-r--r-- 1 root root  426624 Nov 18 22:28 /usr/lib/aarch64-linux-gnu/libGLX_mesa.so.0.0.0
lrwxrwxrwx 1 root root      15 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLX.so -> libGLX.so.0.0.0
lrwxrwxrwx 1 root root      15 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLX.so.0 -> libGLX.so.0.0.0
-rw-r--r-- 1 root root   63880 May 10  2019 /usr/lib/aarch64-linux-gnu/libGLX.so.0.0.0

SaddamBInSyed, 2020-01-27:
@dusty-nv ,

still getting the errors.
```

make
[  1%] Linking CXX shared library ../aarch64/lib/libjetson-utils.so
/usr/bin/ld: cannot find -lGL
collect2: error: ld returned 1 exit status
utils/CMakeFiles/jetson-utils.dir/build.make:2627: recipe for target 'aarch64/lib/libjetson-utils.so' failed
make[2]: *** [aarch64/lib/libjetson-utils.so] Error 1
CMakeFiles/Makefile2:927: recipe for target 'utils/CMakeFiles/jetson-utils.dir/all' failed
make[1]: *** [utils/CMakeFiles/jetson-utils.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

```

dusty-nv, 2020-01-27:
Hmm for some reason, your `libGL.so` is symlink'd to `/usr/lib/aarch64-linux-gnu/tegra/libGL.so`, which doesn't exist on my system.

Try running this:

```bash
sudo ln -s /usr/lib/aarch64-linux-gnu/libGL.so.1 /usr/lib/aarch64-linux-gnu/libGL.so
```
You may also want/need to run `sudo ldconfig` after.

SaddamBInSyed, 2020-01-28:
@dusty-nv  thanks for reply.

when I run the cmd I got the below error.

 sudo ln -s /usr/lib/aarch64-linux-gnu/libGL.so.1 /usr/lib/aarch64-linux-gnu/libGL.so
[sudo] password for deeplearning:
ln: failed to create symbolic link '/usr/lib/aarch64-linux-gnu/libGL.so': File exists

thariqkhalid, 2020-02-06:
you can do 
sudo ln -sfn /usr/lib/aarch64-linux-gnu/libGL.so.1 /usr/lib/aarch64-linux-gnu/libGL.so

It worked for me

SaddamBInSyed, 2020-02-10:
@thariqkhalid 
Thank you. it worked.

huanglidongTX2, 2020-02-13:
I found the same problem on TX2.
when i run"make",it will be :
[ 66%] Built target jetson-utils
[ 66%] Built target jetson-inference
[ 66%] Linking CXX executable ../../aarch64/bin/imagenet-console
/usr/bin/ld: cannot find -lGL
collect2: error: ld returned 1 exit status
examples/imagenet-console/CMakeFiles/imagenet-console.dir/build.make:100: recipe for target 'aarch64/bin/imagenet-console' failed
make[2]: *** [aarch64/bin/imagenet-console] Error 1
CMakeFiles/Makefile2:177: recipe for target 'examples/imagenet-console/CMakeFiles/imagenet-console.dir/all' failed
make[1]: *** [examples/imagenet-console/CMakeFiles/imagenet-console.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

huanglidongTX2, 2020-02-13:
And I have tried to use the command"sudo ln -sfn /usr/lib/aarch64-linux-gnu/libGL.so.1 /usr/lib/aarch64-linux-gnu/libGL.so",  but  it did't work

dusty-nv, 2020-02-13:
@huanglidongTX2 , which version of JetPack are you running?   Had you installed or upgraded packages before installing jetson-inference, which may have changed the OpenGL configuration?

You can try installing this:  https://github.com/NVIDIA/libglvnd

huanglidongTX2, 2020-02-13:
My version is Jetpack4.2, I have tried the solution of the website, and it didn't work. Finally I reflashed the TX2 ,and then I did nothing , I just followed your guide:
sudo apt-get update (do not upgrade)
sudo apt-get install git cmake libpython3-dev python3-numpy
git clone --recursive https://github.com/dusty-nv/jetson-inference
.....
and it can be installed successfully,thank you very  much!