dusty-nv/jetson-inference #487: Choosing high accuracy model for Detectnet-camera for Jetson-nano?

SaddamBInSyed, 2020-01-28:
I have bought my second jetson nano board,
after installing jetson-inference SDK while testing detectnet-camera (USB camera) code with default network model then the object detection accuracy is very poor.
I am focusing on detecting vehicles (truck, lorry etc) here. 

1. Is there any high accuracy network model available with this repo?
2. How to use YOLO model in detectnet-camera app?
3. How to pass to IP camera URI to detectnet-camera code?

downloaded model. please advise high accuracy model
Thank you.

[image]

dusty-nv, 2020-01-28:
Have you tried the `ssd-inception-v2` model?   It should be higher accuracy.

Also, if you are focusing on detecting vehicles, try the models that come with DeepStream SDK, as those are trained for urban/street scenarios like detecting vehicles and pedestrians.

SaddamBInSyed, 2020-01-28:
I have tried with deepstream_test_3_  sample code with IP camera  but no luck

I have posted this question long back but no one answered me.

https://devtalk.nvidia.com/default/topic/1070326/deepstream-sdk/error-a-lot-of-buffers-are-being-dropped-when-running-ds-sdk-python-sample-ipcamera-on-nano/post/5422455/#5422455

so I came back and tried jetson-inference (at least it's running without error).

could you please answer me, 

1. How to use YOLO model in detectnet-camera app?
2. How to pass to IP camera URI to detectnet-camera code?

dusty-nv, 2020-01-28:
Sorry, the support team is away on holiday for the Lunar New Year, I’m sure they will get back to you relatively soon about your DeepStream question.

YOLO isn't supported in the jetson-inference library, it would require special support and I've found the SSD-Mobilenet/Inception performance to be faster.  There is however, a YOLO sample now included with TensorRT SDK - you can find it on your Jetson at:   `/usr/src/tensorrt/samples/python`

What kind of IP camera are you using?  Here is an older fork of jetson-inference where GigEVision support was added:   https://github.com/Abaco-Systems/jetson-inference-gv

SaddamBInSyed, 2020-01-29:
>  https://github.com/Abaco-Systems/jetson-inference-gv

while executing make cmd I am getting below error.

`make
[  3%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/cuda/jetson-inference_generated_cudaYUV.cu.o
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
[  7%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/jetson-inference_generated_imageNet.cu.o
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
[ 10%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/cuda/jetson-inference_generated_cudaFont.cu.o
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
[ 14%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/cuda/jetson-inference_generated_cudaNormalize.cu.o
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
[ 17%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/cuda/jetson-inference_generated_cudaOverlay.cu.o
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
/home/svnano/machinelearning/abaco_jetson-inference-gv/cuda/cudaOverlay.cu(29): warning: variable "thick" was declared but never referenced

/home/svnano/machinelearning/abaco_jetson-inference-gv/cuda/cudaOverlay.cu(8): warning: function "eq_less" was declared but never referenced

<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
[ 21%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/cuda/jetson-inference_generated_cudaRGB-NV12.cu.o
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
In file included from /home/svnano/machinelearning/abaco_jetson-inference-gv/cuda/cudaRGB-NV12.cu:5:0:
/usr/local/cuda/include/math_functions.h:54:2: warning: #warning "math_functions.h is an internal header file and must not be used directly.  This file will be removed in a future CUDA release.  Please use cuda_runtime_api.h or cuda_runtime.h instead." [-Wcpp]
 #warning "math_functions.h is an internal header file and must not be used directly.  This file will be removed in a future CUDA release.  Please use cuda_runtime_api.h or cuda_runtime.h instead."
  ^~~~~~~
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
In file included from /home/svnano/machinelearning/abaco_jetson-inference-gv/cuda/cudaRGB-NV12.cu:5:0:
/usr/local/cuda/include/math_functions.h:54:2: warning: #warning "math_functions.h is an internal header file and must not be used directly.  This file will be removed in a future CUDA release.  Please use cuda_runtime_api.h or cuda_runtime.h instead." [-Wcpp]
 #warning "math_functions.h is an internal header file and must not be used directly.  This file will be removed in a future CUDA release.  Please use cuda_runtime_api.h or cuda_runtime.h instead."
  ^~~~~~~
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
In file included from /home/svnano/machinelearning/abaco_jetson-inference-gv/cuda/cudaRGB-NV12.cu:5:0:
/usr/local/cuda/include/math_functions.h:54:2: warning: #warning "math_functions.h is an internal header file and must not be used directly.  This file will be removed in a future CUDA release.  Please use cuda_runtime_api.h or cuda_runtime.h instead." [-Wcpp]
 #warning "math_functions.h is an internal header file and must not be used directly.  This file will be removed in a future CUDA release.  Please use cuda_runtime_api.h or cuda_runtime.h instead."
  ^~~~~~~
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
In file included from /home/svnano/machinelearning/abaco_jetson-inference-gv/cuda/cudaRGB-NV12.cu:5:0:
/usr/local/cuda/include/math_functions.h:54:2: warning: #warning "math_functions.h is an internal header file and must not be used directly.  This file will be removed in a future CUDA release.  Please use cuda_runtime_api.h or cuda_runtime.h instead." [-Wcpp]
 #warning "math_functions.h is an internal header file and must not be used directly.  This file will be removed in a future CUDA release.  Please use cuda_runtime_api.h or cuda_runtime.h instead."
  ^~~~~~~
[ 25%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/cuda/jetson-inference_generated_cudaResize.cu.o
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
[... cut ...]
      mergeRect( rects[z], make_float6(x1, y1, x2, y2, coverage, z) );
      ^~~~~~~~~
CMakeFiles/jetson-inference.dir/build.make:125: recipe for target 'CMakeFiles/jetson-inference.dir/detectNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/detectNet.cpp.o] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

`

please advise whether Abaco repo will support Jetson nano board?

dusty-nv, 2020-01-29:
> please advise whether Abaco repo will support Jetson nano board?

As mentioned, this was an older fork, so you may want to file an issue on the Abaco repo GitHub or pull out the relevant IP camera code from it:

https://github.com/Abaco-Systems/jetson-inference-gv/blob/master/camera/gvStream.h
https://github.com/Abaco-Systems/jetson-inference-gv/blob/master/camera/rtpStream.h

SaddamBInSyed, 2020-02-11:
> Have you tried the `ssd-inception-v2` model? It should be higher accuracy.
> 
> Also, if you are focusing on detecting vehicles, try the models that come with DeepStream SDK, as **those are trained for urban/street scenarios like detecting vehicles and pedestrians.**

@dusty-nv 

I converted yolov3 onnx model into the tensor RT (.rtf)  model and I can get max up to 4FPS only on my nano. [(instruction here)](https://github.com/jkjung-avt/tensorrt_demos#ssd)

So I am looking for an Nvidia pre-trained vehicle detection model that can be converted to tensor RT the same as above.

please advise.

SaddamBInSyed, 2020-02-12:
> Have you tried the `ssd-inception-v2` model? It should be higher accuracy.
> 
> Also, if you are focusing on detecting vehicles, **try the models that come with DeepStream SDK, as those are trained for urban/street scenarios like detecting vehicles and pedestrians.**

@dusty-nv ,

You mean Can I use that model(font bolded above) into my detectnet_camera code?
Am I correct? 

please advise.

SaddamBInSyed, 2020-02-13:
@dusty-nv 

help appreciated.