dusty-nv/jetson-inference #488: Jetson nano gets power off itself!

SaddamBInSyed, 2020-01-29:
Hi @dusty-nv ,

since 2 days my second jetson nano board going to power off itself.
now when I run detectnet-camera app the board suddenly gets power off itself.

I can provide you the logs if you want to run any cmd on the board.

please advise.

kind note:
I didnt face any similar issue in my first jetson nano board.

dusty-nv, 2020-01-29:
If you run the board in 5W mode (`sudo nvpmodel -m 1`), does it still suddenly power off?

If not, it points to a power supply issue, and you may want to try one of the adapters from this thread:
https://devtalk.nvidia.com/default/topic/1048640/jetson-nano/power-supply-considerations-for-jetson-nano-developer-kit/

SaddamBInSyed, 2020-01-30:
> If you run the board in 5W mode (sudo nvpmodel -m 1), does it still suddenly power off?
it s not happening all the time but sometimes.

regarding the power supply, I am using the same which I already using for another jetson where I didn't face  similar issues till now,

thanks for your reply.

closed 2020-01-30 by SaddamBInSyed

azyslzp, 2020-07-24:
I also meet the same problem! When I run the detectnet.py, the nano suddenly gets power off itself.

dusty-nv, 2020-07-24:
> I also meet the same problem! When I run the detectnet.py, the nano suddenly gets power off itself.

Hi @azyslzp , try running your board in 5W mode:   `sudo nvpmodel -m 1`

If it no longer powers off, it means your power supply isn't consistently delivering enough current for the 10W mode.

azyslzp, 2020-07-25:
Thanks! It is worked after changing the DC power to 5V-4A