dusty-nv/jetson-inference #491: Problem imagenet-camera.py

Alex-259, 2020-01-31:
Hello, 

i want to try the Jetson Nano with my Logitech C920 USB Cam.
When i type `v4l2-ctl -d /dev/video0 --list-formats-ext` this is showed to me:

ioctl: VIDIOC_ENUM_FMT
Index       : 0
       Type        : Video Capture
        Pixel Format: 'YUYV'
        Name        : YUYV 4:2:2
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 160x90
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 160x120
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 176x144
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 320x180
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 432x240
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
[... 342 lines cut ...]
[TRT]   Post-Process  CPU   0.33928ms  CUDA   0.33854ms
[TRT]   Total         CPU  20.64385ms  CUDA  13.08552ms
[TRT]   ------------------------------------------------

[TRT]   note -- when processing a single image, run 'sudo jetson_clocks' before
                to disable DVFS for more accurate profiling/timing measurements

[cuda]  cudaGetLastError()
[cuda]     unknown error (error 30) (hex 0x1E)
[cuda]     /home/geng/jetson-inference/utils/cuda/cudaRGB.cu:76
[cuda]  cudaRGB8ToRGBA32((uchar3*)input, (float4*)mRGBA[mLatestRGBA], mWidth, mHeight)
[cuda]     unknown error (error 30) (hex 0x1E)
[cuda]     /home/geng/jetson-inference/utils/camera/gstCamera.cpp:275
[gstreamer] gstCamera failed to convert frame to RGBA
Traceback (most recent call last):
  File "imagenet-camera.py", line 57, in <module>
    img, width, height = camera.CaptureRGBA()
Exception: jetson.utils -- gstCamera failed to CaptureRGBA()
PyTensorNet_Dealloc()
jetson.utils -- PyFont_Dealloc()
jetson.utils -- PyCamera_Dealloc()
[gstreamer] closing gstCamera for streaming, transitioning pipeline to GST_STATE_NULL
jetson.utils -- PyDisplay_Dealloc()

I don't know how to fix this. Have anybody an idea?

gasparramoa, 2020-02-17:
Hi!!! Probably is because you don't have the correct width and height specified for the device 
> Size: Discrete 640x480

By default, the discrete size is 1280 * 720.
> v4l2src device=/dev/video0 ! video/x-raw, width=(int)1280, height=(int)720, format=YUY2 !

Just specify the discrete size like this: 
`python3 imagenet-camera.py --camera=/dev/video0 --width=640 --heigh=480`

Alex-259, 2020-02-17:
Hi gasparramoa,

thanks for your answer.
Unfortunately this results in the same error...

I have no idea what i can change.

gasparramoa, 2020-02-17:
> Hi gasparramoa,
> 
> thanks for your answer.
> Unfortunately this results in the same error...
> 
> I have no idea what i can change.

Sorry, the height parameter was badly written (heigh - height), don't know if you notice it.

Just do:
`python3 imagenet-camera.py --camera=/dev/video0 --width=640 --height=480`

Alex-259, 2020-02-21:
Yes I have noticed this.
Unfortunately it has the same error message.

I have tried also to convert the images in RGBAf (float) images but the code results in the same error :/

closed 2020-03-16 by Alex-259