dusty-nv/jetson-inference #492: About convert tf model to uff model

DoriHp, 2020-02-01:
I wanna use another custom ssd-mobilenet v1 model instead of coco model. Before that, I tried to convert this model to uff format by this command: `python3 /usr/lib/python3.6/dist-packages/uff/bin/convert_to_uff.py frozen_inference_graph_face.pb -o face_trt.uff  -O NMS -p config.py`. A new uff file was produced. But when I replace this file with your file, It din't work:
`[TRT]   UffParser: Validator error: FeatureExtractor/MobilenetV1/zeros_4: Unsupported operation _Fill
[TRT]   failed to parse UFF model '/usr/local/bin/networks/SSD-Mobilenet-v1/ssd_mobilenet_v1_coco.uff'
[TRT]   device GPU, failed to load networks/SSD-Mobilenet-v1/ssd_mobilenet_v1_coco.uff
detectNet -- failed to initialize.
`
Here is the model that I tried to convert:
https://github.com/yeephycho/tensorflow-face-detection/blob/master/model/frozen_inference_graph_face.pb
Here is my config file:
https://1drv.ms/u/s!AoWJ24OtfQT1ge5KaNFvcxh3E5dEpA?e=xoMImJ
Can you show me what I did wrong? Or suggest me your way to do this task?
Thank in advance!

bobtattvamasi, 2020-05-12:
Same problem :/

bobtattvamasi, 2020-05-12:
Did you solve it?

leandrovrabelo, 2020-05-12:
I also stopped on this step, I have my custom SSD model and I would like to use it on DetecNet.

DoriHp, 2020-05-19:
To use your custom ssd model, you must rollback TF Object Detection API to old commits (early 2018) since new version of this API have changed some layers in model's architecture and convert it use above config file or fix config file to adapted to that layer. I tried the first way and I work well.

danhnguyen12091998, 2020-06-17:
@DoriHp Can you show me the detailed instructions because I have the same issue with you :(

DoriHp, 2020-06-18:
@danhnguyen12091998  For now if you still want to use mobilenet-ssd with tensorrt intergration on Jetson platform, I think you should try this (repo)[https://github.com/ceccocats/tkDNN#4export-weights-for-mobilenetssd], of course in case you're familiar with C++ (:3). If you still want to use jetson-inference, as I described above, you should rollback Tensorflow Object Detection API to commit at early 2018 (I don't remember exactly), but you must have cuda 9.0 and a proper tensorflow version. After that, let trains your own ssd model with my config file (you can customize it, of course) and follow steps that I presented. I haven't re-use this repo for a long time so I can
t remember more detailed things.

closed 2020-06-18 by DoriHp

danhnguyen12091998, 2020-06-18:
@DoriHp you mean I need to train the model with the old commit of Object Detection API, right ?

leandrovrabelo, 2020-06-18:
Hi @danhnguyen12091998 , you may try [these](https://github.com/dusty-nv/jetson-inference/issues/585#issuecomment-636595912) steps to see if the convertion works.

You also can try to adapt [my code](https://github.com/leandrovrabelo/object_detection/tree/master/uff_convert) to your needs.

I made a work around for the inference to work on DetecNet, i'm still trying to figure out how to make my model works on its own base.

DoriHp, 2020-06-18:
@danhnguyen12091998 you're right. This repo is only compatible with model that was trained on old commit of Object Detection API. Maybe other ways're still exist, but I only know this way.

danhnguyen12091998, 2020-06-18:
@DoriHp I found the model was trained in 2017, but when I build engine, it has a same issues :( 
I found it in this link: https://towardsdatascience.com/how-to-train-a-tensorflow-face-object-detection-model-3599dcd0c26f

DoriHp, 2020-06-18:
As I told, I haven't re-used this repo for a long time after I succeed getting my custom model. I'will try to reproduce my work soon. In that time, you must try by yourself or find another possible way. And as I described, tkDNN is very hopeful.

danhnguyen12091998, 2020-06-18:
I need to use this model for jetson - inference, so I think tkDNN is not suitable for me. Thanks for your support @DoriHp

DoriHp, 2020-06-27:
@danhnguyen12091998  are you still here :). I reproduced my work recently, and create a repo based on TF object detection API at [here](https://github.com/DoriHp/forked-TF-Object-Detection). Please follow the instruction carefully, if something get wrong, feel free to contact me :))

danhnguyen12091998, 2020-06-28:
Many thanks for your support @DoriHp !! I will follow your instruction carefully. But I hope I can add your Skype for easily to contact.