dusty-nv/jetson-inference #495: How to modify my-detection.py for my own dataset?

AK51, 2020-02-02:
I have done the "Collecting your own Datasets" and it can classify the object. Cool.
May I know how to change the function call of my-detection.py?

# load the object detection model
net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)

Changing ssd-mobilenet-v2 to my onnx file? Thx

AK51, 2020-02-08:
After Collecting your own Datasets and it can classify my own objects. Then, how to 
"Performing object detection to locate object coordinates"?
What tools to capture the data pictures and draw a box around it?
What is the procedure to create the onnx and put into the detectnet for example? Thx

dusty-nv, 2020-02-18:
Hi @AK51, please see this post for suggestions of re-training object detection model on customized datasets:

https://devtalk.nvidia.com/default/topic/1070225/jetson-nano/digits-or-somthing-else/post/5421938/#5421938

GOBish, 2020-07-08:
Hi @dusty-nv , any chance you can add to your excellent tutorial a section on training a custom detectnet object detection model, similar to the one you did for object classification? I have not found another tutorial that walks through the process clearly.

dusty-nv, 2020-07-08:
Hi @GOBish, I am working on it now actually, I hope it will be ready soon.

The PyTorch training code that it uses is here:  https://github.com/dusty-nv/pytorch-ssd

Indeed custom object detection training has not been very simple in the past, however this PyTorch way is pretty easy because the model can be exported to ONNX and then imported to TensorRT.

GOBish, 2020-07-08:
thank you @dusty-nv ! I will try the code you reference, and I really look forward to your tutorial!

dusty-nv, 2020-07-12:
@GOBish, alpha version of the tutorial is here:  https://github.com/dusty-nv/jetson-inference/blob/dev/docs/pytorch-ssd.md

Let me know how it goes - thanks.

GOBish, 2020-07-13:
Wow, this is awesome @dusty-nv . Thank you for all your hard work. I am really looking forward to the next step, "collecting your own datasets" as well. That will be a huge help for my work.

Thanks again.

dusty-nv, 2020-07-13:
> I am really looking forward to the next step, "collecting your own datasets" as well

Yep working on that part now...will let you know when it's ready to try.

dusty-nv, 2020-07-15:
@GOBish - see here:  https://github.com/dusty-nv/jetson-inference/blob/dev/docs/pytorch-collect-detection.md

You will need to pull the latest for some updates that went into the camera-capture tool.

GOBish, 2020-07-18:
Running through the tutorial today, will let you know how it goes!! Very much appreciate your hard work on this.

GOBish, 2020-07-18:
Hi @dusty-nv — would you recommend a method for putting negative images in the mode to lower false detections?

GOBish, 2020-07-18:
awesome work on this, @dusty-nv .
One recomendation for the tutorial on using your own images for object detection training --
more guidance on folder structure and file needs if you are using your own images and not using the camera tool (which is excellent, by the way).
Right now I have to use the camera tool to see the file structure it creates (i.e. Annotations, ImageSets, JPEGImages) and the files (i.e. trainval.txt) so that I can know where to put the images, and the xml files from labelimg.
I think adding some guidance for that would be useful.

Thanks again for the great work on this.

GOBish, 2020-07-19:
I'm having a tough time getting the object detection training to run on my system.

1. When I try training using images captured and labelled on the jetson using the camera app, I get the following output and error:

```
~/jetson-inference/python/training/detection/ssd$ python3 train_ssd.py --dataset-type=voc --data=/home/gbenel/Desktop/data_test/ --model-dir=/home/gbenel/Desktop/data_test/ 
2020-07-18 20:19:47 - Using CUDA...
2020-07-18 20:19:47 - Namespace(balance_data=False, base_net=None, base_net_lr=0.001, batch_size=4, checkpoint_folder='/home/gbenel/Desktop/data_test/', dataset_type='voc', datasets=['/home/gbenel/Desktop/data_test/'], debug_steps=10, extra_layers_lr=None, freeze_base_net=False, freeze_net=False, gamma=0.1, lr=0.01, mb2_width_mult=1.0, milestones='80,100', momentum=0.9, net='mb1-ssd', num_epochs=30, num_workers=2, pretrained_ssd='models/mobilenet-v1-ssd-mp-0_675.pth', resume=None, scheduler='cosine', t_max=100, use_cuda=True, validation_epochs=1, weight_decay=0.0005)
2020-07-18 20:19:47 - Prepare training datasets.
2020-07-18 20:19:47 - VOC Labels read from file: ('BACKGROUND', 'BACKGROUND', 'BACKGROUND', 'BACKGROUND', 'BACKGROUND', 'RedGate', 'BlueGate')
2020-07-18 20:19:47 - Stored labels into file /home/gbenel/Desktop/data_test/labels.txt.
2020-07-18 20:19:47 - Train dataset size: 2
2020-07-18 20:19:47 - Prepare Validation datasets.
2020-07-18 20:19:47 - VOC Labels read from file: ('BACKGROUND', 'BACKGROUND', 'BACKGROUND', 'BACKGROUND', 'BACKGROUND', 'BACKGROUND', 'RedGate', 'BlueGate')
2020-07-18 20:19:47 - Validation dataset size: 3
2020-07-18 20:19:47 - Build network.
2020-07-18 20:19:47 - Init from pretrained ssd models/mobilenet-v1-ssd-mp-0_675.pth
2020-07-18 20:19:47 - Took 0.17 seconds to load the model.
2020-07-18 20:19:51 - Learning rate: 0.01, Base net learning rate: 0.001, Extra Layers learning rate: 0.01.
2020-07-18 20:19:51 - Uses CosineAnnealingLR scheduler.
2020-07-18 20:19:51 - Start training from epoch 0.
/home/gbenel/.local/lib/python3.6/site-packages/torch/optim/lr_scheduler.py:123: UserWarning: Detected call of `lr_scheduler.step()` before `optimizer.step()`. In PyTorch 1.1.0 and later, you should call them in the opposite order: `optimizer.step()` before `lr_scheduler.step()`.  Failure to do this will result in PyTorch skipping the first value of the learning rate schedule. See more details at https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate
  "https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate", UserWarning)
/home/gbenel/jetson-inference/python/training/detection/ssd/vision/transforms/transforms.py:247: VisibleDeprecationWarning: Creating an ndarray from ragged nested sequences (which is a list-or-tuple of lists-or-tuples-or ndarrays with different lengths or shapes) is deprecated. If you meant to do this, you must specify 'dtype=object' when creating the ndarray
  mode = random.choice(self.sample_options)
/home/gbenel/.local/lib/python3.6/site-packages/torch/nn/_reduction.py:44: UserWarning: size_average and reduce args will be deprecated, please use reduction='sum' instead.
  warnings.warn(warning.format(ret))
/media/nvidia/WD_NVME/PyTorch/JetPack_4.4/GA/pytorch-v1.6.0-rc2/aten/src/THCUNN/ClassNLLCriterion.cu:108: cunn_ClassNLLCriterion_updateOutput_kernel: block: [0,0,0], thread: [12,0,0] Assertion `t >= 0 && t < n_classes` failed.
/media/nvidia/WD_NVME/PyTorch/JetPack_4.4/GA/pytorch-v1.6.0-rc2/aten/src/THCUNN/ClassNLLCriterion.cu:108: cunn_ClassNLLCriterion_updateOutput_kernel: block: [0,0,0], thread: [13,0,0] Assertion `t >= 0 && t < n_classes` failed.
/media/nvidia/WD_NVME/PyTorch/JetPack_4.4/GA/pytorch-v1.6.0-rc2/aten/src/THCUNN/ClassNLLCriterion.cu:108: cunn_ClassNLLCriterion_updateOutput_kernel: block: [0,0,0], thread: [15,0,0] Assertion `t >= 0 && t < n_classes` failed.
/media/nvidia/WD_NVME/PyTorch/JetPack_4.4/GA/pytorch-v1.6.0-rc2/aten/src/THCUNN/ClassNLLCriterion.cu:108: cunn_ClassNLLCriterion_updateOutput_kernel: block: [0,0,0], thread: [16,0,0] Assertion `t >= 0 && t < n_classes` failed.
/media/nvidia/WD_NVME/PyTorch/JetPack_4.4/GA/pytorch-v1.6.0-rc2/aten/src/THCUNN/ClassNLLCriterion.cu:108: cunn_ClassNLLCriterion_updateOutput_kernel: block: [0,0,0], thread: [19,0,0] Assertion `t >= 0 && t < n_classes` failed.
Traceback (most recent call last):
  File "train_ssd.py", line 346, in <module>
    val_loss, val_regression_loss, val_classification_loss = test(val_loader, net, criterion, DEVICE)
  File "train_ssd.py", line 159, in test
    regression_loss, classification_loss = criterion(confidence, locations, labels, boxes)
  File "/home/gbenel/.local/lib/python3.6/site-packages/torch/nn/modules/module.py", line 722, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/gbenel/jetson-inference/python/training/detection/ssd/vision/nn/multibox_loss.py", line 43, in forward
    predicted_locations = predicted_locations[pos_mask, :].reshape(-1, 4)
RuntimeError: copy_if failed to synchronize: cudaErrorAssert: device-side assert triggered

```

2. When I try running using a dataset I created using labelImage (and all folder structure is correct with Annotations, ImageSets, JPEGImages, and text files) I get the following output and error:
```

~/jetson-inference/python/training/detection/ssd$ python3 train_ssd.py --dataset-type=voc --data=/home/gbenel/Desktop/downloadedImages/ --model-dir=/home/gbenel/Desktop/downloadedImages/ 
2020-07-18 20:20:40 - Using CUDA...
[... cut ...]
  File "/home/gbenel/jetson-inference/python/training/detection/ssd/vision/datasets/voc_dataset.py", line 24, in __init__
    self.ids = VOCDataset._read_image_ids(image_sets_file)
  File "/home/gbenel/jetson-inference/python/training/detection/ssd/vision/datasets/voc_dataset.py", line 87, in _read_image_ids
    for line in f:
  File "/usr/lib/python3.6/codecs.py", line 321, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte

```

GOBish, 2020-07-20:
@dusty-nv , this is awesome work. I trained a model based on images I labelled in LabelImage and it is performing well, a great start. I need to add more images (currently using ~500 and 2 classes).

One question - I can run using both detectnet and detectnet.py using the flags as in the demo.
Would you be able to show how the model can be input to my-detection.py that we developed during the tutorial? I can't seem to get it to use the onnx file successfully. It would be useful for me to be able to use this simple script so that I can integrate the object detection into other applications I am developing.

Again, thanks for the great work!

dusty-nv, 2020-07-20:
@GOBish, glad you got it working.  If you look at `detectnet.py`, you can see that it passes the raw command line to `detectNet` when loading the network.  Those custom model parameters are then parsed from the command line by detectNet.

So you can either add the command line argument to `my-detection.py` like `detectnet.py` has, or if you want to hard-code it, you can construct the command line array in code and pass that to detectNet (not ideal I know, but should work).

GOBish, 2020-07-20:
Thanks @dusty-nv .
Even with putting in the full paths for the model, labels, and output files, I still seem to need to run my-detection.py from inside the /python/training/detection/ssd directory. If I run it from anywhere else, the image comes up and then closes after a couple of seconds (I imagine maybe when it makes a detection). Do you know why this is the case?

dusty-nv, 2020-07-20:
It should automatically create the GL window unless the `--headless` flag is specified, or if you are running `detectnet-console.py`

GOBish, 2020-07-20:
It's strange, it creates a GL window without specifying a save file when I use the default model (ssd-mobilenet-v2), but not when I use the onnx model.

dusty-nv, 2020-07-20:
Hmm can you provide the command line you are launching with, so I can try something similar to debug?  Thanks.

GOBish, 2020-07-20:
`gbenel@gbenel:~/jetson-inference/python/examples$ python3 detectnet.py --network=/home/gbenel/jetson-inference/python/training/detection/ssd/models/gates/ssd-mobilenet.onnx --labels=/home/gbenel/jetson-inference/python/training/detection/ssd/models/gates/labels.txt  --input-blob=input_0 --output-cvg=scores --output-bbox=boxes /dev/video2 /home/gbenel/jetson-inference/python/training/detection/ssd/test_gates/output.mp4 --threshold=.2 
`

dusty-nv, 2020-07-23:
Thanks @GOBish , I just patched a bug in the command-line parsing that fixes this.

If you `git pull origin master` on your jetson-inference, and then run `make` and `sudo make install`, it should be fixed.

GOBish, 2020-07-23:
hmm, I still seem to have the same behavior after the git pull, still need to be saving the input video or camera image in order for it to come up on screen.

dusty-nv, 2020-07-23:
Hmm can you post the terminal log?   Are you sure you did a `sudo make install`?   (otherwise python would still run the previous build of the libraries)

What happens if you run it with `display://0` instead of the output video?

```bash
python3 detectnet.py --network=/home/gbenel/jetson-inference/python/training/detection/ssd/models/gates/ssd-mobilenet.onnx --labels=/home/gbenel/jetson-inference/python/training/detection/ssd/models/gates/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes /dev/video2 display://0
```

GOBish, 2020-07-23:
yes, did the sudo make install.

the display://0 got it to work!

GOBish, 2020-07-23:
Do you have any examples if we wanted to run the code with a TensorRT model, not an onnx model?  The basic code example get the pretrained models but I am not sure how to feed it a different trt model (.pb file). I do have the .pbtxt and config files as well for that model.

dusty-nv, 2020-07-23:
> Do you have any examples if we wanted to run the code with a TensorRT mode

Do you mean TensorFlow model (the jetson-inference repo already uses TensorRT)?

The default SSD-Mobilenet model (COCO) is actually a TensorFlow model.   However it isn't all that easy to convert it, which is another positive of using the PyTorch SSD-Mobilenet.  You can see here for how it was converted from TF though:   https://github.com/AastaNV/TRT_object_detection

GOBish, 2020-07-23:
I have converted a tensorflow model I made (transfer off of sad-mobile net-v2) successfully to tensorRT, now I am just looking to run inference with it using the tutorial code.

dusty-nv, 2020-07-23:
There are some extra parameters needed for TensorFlow UFF detection model (like input dims) that aren't parsed from the command line - see where they are defined in the code here:

https://github.com/dusty-nv/jetson-inference/blob/62cec4054cf14b61fb7b722dad91077c57a80bec/c/detectNet.cpp#L258

essentially add an enum and your model info to another `else if` statement there.

GOBish, 2020-07-30:
Hi Dusty, I found that line, but still not sure exactly how to do this -
I have the trt model I have created that is a .pb file, and I have the .pbtxt labels file. I see that detectNet.cpp is looking for a .uff file.

dusty-nv, 2020-08-03:
@GOBish see here for a tool that converts object detection model from TB .pb to .uff - https://github.com/AastaNV/TRT_object_detection

These extra steps are one reason I prefer to just use the PyTorch ONNX workflow.

closed 2020-11-22 by AK51