dusty-nv/jetson-inference #496: How can i use a external camera? (Blaster)

mjack3, 2020-02-05:
Hello, 

I would like to use this repository for make inferences in a jetson tx2 with a blaster camera. In fact, this is my first time working with a jetson so i have a little bit of confusion.

Actually, i decided to install pypylon, which is the recommended software but i dont know really how to do it work with the examples of "create your own object detection program" (using python).

dusty-nv, 2020-02-18:
For others reading this issue, please refer to the solution posted here:

https://devtalk.nvidia.com/default/topic/1071288/jetson-tx2/how-make-inference-with-basler-camera/post/5427899/#5427899

closed 2020-02-18 by dusty-nv

jvitordm, 2022-12-16:
Hi ther @mjack3 , could you please share the solution?

Repository below is not available anymore, thanks!
https://github.com/mjack3/jetson-inference/blob/master/detectnet-basler3.py