dusty-nv/jetson-inference #503: Can you give me a example CMakeLists.txt for imagenet-camera

huanglidongTX2, 2020-02-14:
Recently  I tried to run the example in the folder /jetson-inference/examples/imagenet-camera/, but it can't run successfully. My device is TX2 , ubuntu 16.04, jetpack3.2.
 Here is what i wrote:

**CMakeLists.txt:**

cmake_minimum_required(VERSION 2.8)
project(imagenet-camera)
find_package(jetson-utils)
find_package(jetson-inference)
file(GLOB imagenetCameraSources *.cpp)
file(GLOB imagenetCameraIncludes *.h )
include_directories(/usr/include/tbb  /usr/local/include/jetson-utils  /usr/local/include/jetson-inference )

find_package(CUDA)
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -Wno-write-strings")
cuda_add_executable(imagenet-camera imagenet-camera.cpp)
target_link_libraries(imagenet-camera jetson-inference)
install(TARGETS imagenet-camera DESTINATION bin)

**.cpp**
I don't  alter the .cpp file

I run "cmake ." and "make" in the terminal, then errors came:

nvidia@jetson-1422719019466:~/111$ make 
[ 50%] Building CXX object CMakeFiles/ss.dir/main.cpp.o
In file included from /home/nvidia/111/main.cpp:23:0:
/home/nvidia/111/gstCamera.h:271:9: error: ‘GstFlowReturn’ does not name a type static GstFlowReturn onPreroll(_GstAppSink* sink, void* user_data);
         ^
/home/nvidia/111/gstCamera.h:272:9: error: ‘GstFlowReturn’ does not name a type static GstFlowReturn onBuffer(_GstAppSink* sink, void* user_data);
         ^
/home/nvidia/111/gstCamera.h:283:2: error: ‘_GstBus’ does not name a type  _GstBus*     mBus;
  ^
/home/nvidia/111/gstCamera.h:285:2: error: ‘_GstElement’ does not name a type  _GstElement* mPipeline;
  ^
CMakeFiles/ss.dir/build.make:62: recipe for target 'CMakeFiles/ss.dir/main.cpp.o' failed
make[2]: *** [CMakeFiles/ss.dir/main.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ss.dir/all' failed
make[1]: *** [CMakeFiles/ss.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

I don't konw how to solve it ,plese help me ,
can you give me a CMakeLists.txt example for it,  thank you very much!

dusty-nv, 2020-02-18:
Hi @huanglidongTX2 , you might want to try adding this line to your CMakeLists.txt:

`include_directories(/usr/include/gstreamer-1.0 /usr/lib/aarch64-linux-gnu/gstreamer-1.0/include /usr/include/glib-2.0 /usr/include/libxml2 /usr/lib/aarch64-linux-gnu/glib-2.0/include/)`

huanglidongTX2, 2020-02-20:
ok ,it works ,thank you very much!